android/camera-samples #212: Preview issues in camera-view alpha09 & alpha10

mgarciaguerrero, 2020-04-24:
### Description
I'm facing some issues of your sample app when running on my testing device. The problem is that the preview is not fully displayed in screen (screenshot below). 

After some investigation, I can confirm it happens in both camera-view versions `alpha09` and `alpha10` (`alpha08` works fine). If I set the `preferredImplementationMode` to use `TextureView` (as recommended in the [documentation](https://developer.android.com/jetpack/androidx/releases/camera#camera-view-1.0.0-alpha09)), then the issue disappears. 
```kotlin
val camera = cameraProvider.bindToLifecycle(lifecycleOwner, cameraSelector, preview)
previewView.preferredImplementationMode = ImplementationMode.TEXTURE_VIEW
preview.setSurfaceProvider(previewView.createSurfaceProvider(camera.cameraInfo))
````
However, taking into account this comment, seems that is it not recommended:

- https://github.com/android/camera-samples/pull/198#discussion_r406349236

### Screenshot
[image]

### Device information
- **Device name**: Samsung Galaxy Note5
- **Model**: SM-N920C
- **Android version**: 7

TheMelody, 2020-04-28:
previewView.preferredImplementationMode = ImplementationMode.TEXTURE_VIEW

Great, solved the official bug。

mgarciaguerrero, 2020-04-28:
@owahltinez any advice on this? Is `previewView.preferredImplementationMode = ImplementationMode.TEXTURE_VIEW` the official solution to solve this issue?

owahltinez, 2020-04-28:
Forcing `TextureView` implementation is only a temporary workaround, which I would discourage for production apps. The long-term plan is for the CameraX library to choose the appropriate implementation for all devices. I reported the issue in the library's tracker, and I'm closing it here since it's not an issue with the sample (but feel free to continue the conversation).

closed 2020-04-28 by owahltinez

mgarciaguerrero, 2020-04-28:
Thanks @owahltinez for the clarification 👍

clhols, 2020-05-19:
@owahltinez Can you provide a link to the issue you created on the library's tracker, so we can monitor when this is fixed?