android/camera-samples #214: Retrieving bitmap from OnImageCapturedCallback not working since beta02

mlilienberg, 2020-04-26:
Since beta02 our way to retrieve the bitmap within OnImageCapturedCallback is not working anymore. Either it is because we do something wrong, than an example could be helpful or it is broken in beta02 and beta03.

This is what we do in onCaptureSuccess:

```
@SuppressLint("UnsafeExperimentalUsageError")
override fun onCaptureSuccess(image: ImageProxy) {
   val rotation = when(image.imageInfo.rotationDegrees) {
      90 -> FirebaseVisionImageMetadata.ROTATION_90
      180 -> FirebaseVisionImageMetadata.ROTATION_180
      270 -> FirebaseVisionImageMetadata.ROTATION_270
      else -> FirebaseVisionImageMetadata.ROTATION_0
   }
   val bitmap = FirebaseVisionImage.fromMediaImage(image.image!!, rotation).bitmap
   super.onCaptureSuccess(image)
}
```

I tried multiple approaches to decode the buffer, but all are crashing with similar exceptions like
`Attempt to invoke virtual method 'int android.graphics.Bitmap.getWidth()'`

When using 0 as rotation at least the call `FirebaseVisionImage.fromMediaImage(image.image!!, rotation)` succeeds, however the bitmap is still null.

Does this Callback require some additional steps in configuration?

owahltinez, 2020-04-27:
This repo is for issues with the samples. For questions related to API usage, please use the `android-camerax` tag in StackOverflow or use the [developer forums](https://g.co/camerax/developers).

closed 2020-04-27 by owahltinez

nilknarfuw, 2020-05-12:
This is a bug that is present in beta02 and beta03 of CameraX https://issuetracker.google.com/153249512. The change has already been merged but won't be available until beta04 is released. 
Also linking associated buganizer ticket future reference https://issuetracker.google.com/155109625