android/camera-samples #215: java.lang.RuntimeException: setVideoFrameRate failed

ZhuSniffsTheRose, 2020-04-27:
at android.media.MediaRecorder.setVideoFrameRate(Native Method)
        at com.example.android.camera2.video.fragments.CameraFragment.createRecorder(CameraFragment.kt:227)
        at com.example.android.camera2.video.fragments.CameraFragment.access$createRecorder(CameraFragment.kt:72)
        at com.example.android.camera2.video.fragments.CameraFragment$recorderSurface$2.invoke(CameraFragment.kt:108)
        at com.example.android.camera2.video.fragments.CameraFragment$recorderSurface$2.invoke(CameraFragment.kt:72)
        at kotlin.SynchronizedLazyImpl.getValue(LazyJVM.kt:74)
        at com.example.android.camera2.video.fragments.CameraFragment.getRecorderSurface(Unknown Source:2)
        at com.example.android.camera2.video.fragments.CameraFragment.access$getRecorderSurface$p(CameraFragment.kt:72)
        at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1.invokeSuspend(CameraFragment.kt:246)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at android.os.Handler.handleCallback(Handler.java:793)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:173)
        at android.app.ActivityThread.main(ActivityThread.java:6698)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:782)

owahltinez, 2020-04-27:
Thanks for reporting! Please answer a few questions to help us find the root cause of this issue:
* Which sample app are you using?
* Did you modify the sample app in any way?
* Can you provide concrete steps for us to reproduce this issue?
* What device and Android version are you using for testing?

ZhuSniffsTheRose, 2020-04-28:
1. Sample is  Camera2Video
2. Not modify, just clone and run Camera2Video
3.Run Camera2Video and  click any one of buttons;
4. Device Info: MIUI 10 8.11.22.  Android Version: 8.0.0

evansgelist, 2020-04-29:
same problem!
MIUI 11.0.2
8.1.0

Sultan91, 2020-06-07:
Same problem was on virtual devices <= API 25,
On nexus S API 29 works sort of ok

owahltinez, 2020-06-09:
Thank you @Sultan91, I was able to reproduce the issue on emulator. It looks like there is some precision issue with the way FPS is computed, it should be fixed now.

commit 17c4f2bb (2020-06-09): Change computation of FPS in Camera2Video

- Fixes #215

Change-Id: Ica3de00bd25e1ffd74b8bf088c184969322cb4f7
closed 2020-06-09 by owahltinez