android/camera-samples #216: Video recording is crashing

BrunoAndroidDev, 2020-04-27:
Hello,

The Camera2Video sample is crashing on my side when I try the record feature.
I've testing on 2 devices :
- Samsung Galaxy XCover 4S (SM-G398FN) with Android 9
- Motorola Moto G7 plus (XT1965-3) with Android 10

The crash appears when I tap on the "record" button, even I choose Back or Front Camera. Here is the logs on the Galaxy XCover:

> 2020-04-27 11:57:18.295 19669-19669/com.android.example.camera2.video D/SurfaceView: onWindowVisibilityChanged(0) true com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ......I. 0,0-0,0 #7f0800fa app:id/view_finder} of ViewRootImpl@b2719a8[CameraActivity]
> 2020-04-27 11:57:18.309 19669-19669/com.android.example.camera2.video D/SensorManager: registerListener :: 0, LIS2DS Accelerometer, 200000, 0,  
> 2020-04-27 11:57:18.340 19669-19669/com.android.example.camera2.video D/ViewRootImpl@b2719a8[CameraActivity]: Relayout returned: old=[0,0][720,1280] new=[0,0][720,1280] result=0x1 surface={true 494566207488} changed=false
> 2020-04-27 11:57:18.347 19669-19669/com.android.example.camera2.video D/SurfaceView: show() Surface(name=SurfaceView - com.android.example.camera2.video/com.example.android.camera2.video.CameraActivity@925d767@0[19669])/@0x209d162 com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ......ID 0,0-720,1280 #7f0800fa app:id/view_finder}
> 2020-04-27 11:57:18.354 19669-19669/com.android.example.camera2.video D/SurfaceView: surfaceCreated 1 #8 com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ......ID 0,0-720,1280 #7f0800fa app:id/view_finder}
> 2020-04-27 11:57:18.374 19669-19669/com.android.example.camera2.video D/CameraFragment: View finder size: 720 x 1280
> 2020-04-27 11:57:18.374 19669-19669/com.android.example.camera2.video D/CameraFragment: Selected preview size: 1280x720
> 2020-04-27 11:57:18.375 19669-19669/com.android.example.camera2.video D/SurfaceView: setPackageUsesOwnResolution() (Java SurfaceView) for com.android.example.camera2.video: 1280x720 (true)
> 2020-04-27 11:57:18.376 19669-19669/com.android.example.camera2.video D/SurfaceView: surfaceChanged (720,1280) 1 #8 com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ......ID 0,0-720,1280 #7f0800fa app:id/view_finder}
> 2020-04-27 11:57:18.382 19669-19669/com.android.example.camera2.video D/CameraFragment: Orientation changed: 90
> 2020-04-27 11:57:18.391 19669-19669/com.android.example.camera2.video D/AutoFitSurfaceView: Measured dimensions set: 720 x 1280
> 2020-04-27 11:57:18.393 19669-19669/com.android.example.camera2.video D/AutoFitSurfaceView: Measured dimensions set: 720 x 1280
> 2020-04-27 11:57:18.394 19669-19669/com.android.example.camera2.video D/SurfaceView: show() Surface(name=SurfaceView - com.android.example.camera2.video/com.example.android.camera2.video.CameraActivity@925d767@0[19669])/@0x209d162 com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ........ 0,0-720,1280 #7f0800fa app:id/view_finder}
> 2020-04-27 11:57:18.403 19669-19669/com.android.example.camera2.video D/SurfaceView: surfaceChanged (1280,720) 1 #5 com.example.android.camera.utils.AutoFitSurfaceView{925d767 V.E...... ........ 0,0-720,1280 #7f0800fa app:id/view_finder}
> 2020-04-27 11:57:18.473 19669-19669/com.android.example.camera2.video D/StagefrightPluginLoader: Failed to load library: libstagefright_ccodec.so (dlopen failed: library "libstagefright_ccodec.so" not found)
> 2020-04-27 11:57:18.473 19669-19669/com.android.example.camera2.video D/StagefrightPluginLoader: Handle or CreateInputSurface symbol is null
> 2020-04-27 11:57:18.474 19669-19669/com.android.example.camera2.video I/OMXClient: IOmx service obtained
> 2020-04-27 11:57:18.475 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setup
> 2020-04-27 11:57:18.477 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setAudioSource(1)
> 2020-04-27 11:57:18.480 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setVideoSource(2)
> 2020-04-27 11:57:18.485 19669-19669/com.android.example.camera2.video W/MediaRecorder: setOutputFile: path = /data/user/0/com.android.example.camera2.video/files/VID_2020_04_27_11_57_18_483.mp4
> 2020-04-27 11:57:18.485 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setParameter()
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setVideoFrameRate(30)
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setVideoSize(1152, 864)
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setVideoEncoder(2)
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setAudioEncoder(3)
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: android_media_MediaRecorder_setInputSurface
> 2020-04-27 11:57:18.486 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: get_persistentSurface
> 2020-04-27 11:57:18.490 19669-19669/com.android.example.camera2.video W/MediaRecorder: prepare: mPath = /data/user/0/com.android.example.camera2.video/files/VID_2020_04_27_11_57_18_483.mp4
> 2020-04-27 11:57:18.491 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: setOutputFile
> 2020-04-27 11:57:18.491 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: prepare
> 2020-04-27 11:57:18.577 19669-19669/com.android.example.camera2.video I/MediaRecorderJNI: release
> 2020-04-27 11:57:20.548 19669-19669/com.android.example.camera2.video D/ViewRootImpl@b2719a8[CameraActivity]: ViewPostIme pointer 0
> 2020-04-27 11:57:20.572 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setup
> 2020-04-27 11:57:20.574 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setAudioSource(1)
> 2020-04-27 11:57:20.579 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setVideoSource(2)
> 2020-04-27 11:57:20.582 19669-19774/com.android.example.camera2.video W/MediaRecorder: setOutputFile: path = /data/user/0/com.android.example.camera2.video/files/VID_2020_04_27_11_57_18_483.mp4
> 2020-04-27 11:57:20.582 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setParameter()
> 2020-04-27 11:57:20.582 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setVideoFrameRate(30)
> 2020-04-27 11:57:20.583 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setVideoSize(1152, 864)
> 2020-04-27 11:57:20.583 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setVideoEncoder(2)
> 2020-04-27 11:57:20.583 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setAudioEncoder(3)
> 2020-04-27 11:57:20.583 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: android_media_MediaRecorder_setInputSurface
> 2020-04-27 11:57:20.583 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: get_persistentSurface
> 2020-04-27 11:57:20.584 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setParameter()
> 2020-04-27 11:57:20.586 19669-19774/com.android.example.camera2.video W/MediaRecorder: prepare: mPath = /data/user/0/com.android.example.camera2.video/files/VID_2020_04_27_11_57_18_483.mp4
> 2020-04-27 11:57:20.587 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: setOutputFile
> 2020-04-27 11:57:20.587 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: prepare
> 2020-04-27 11:57:20.669 19669-19669/com.android.example.camera2.video D/ViewRootImpl@b2719a8[CameraActivity]: ViewPostIme pointer 1
> 2020-04-27 11:57:20.676 19669-19776/com.android.example.camera2.video D/CameraFragment: Recording stopped. Output file: /data/user/0/com.android.example.camera2.video/files/VID_2020_04_27_11_57_18_483.mp4
> 2020-04-27 11:57:20.677 19669-19776/com.android.example.camera2.video I/MediaRecorderJNI: stop
> 2020-04-27 11:57:20.677 19669-19776/com.android.example.camera2.video E/MediaRecorder: stop called in an invalid state: 4
> 2020-04-27 11:57:20.703 19669-19774/com.android.example.camera2.video I/MediaRecorderJNI: start
> 2020-04-27 11:57:20.711 19669-19776/com.android.example.camera2.video E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-3
>     Process: com.android.example.camera2.video, PID: 19669
>     java.lang.IllegalStateException
>         at android.media.MediaRecorder._stop(Native Method)
>         at android.media.MediaRecorder.stop(MediaRecorder.java:1370)
[... cut ...]
>         at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
> 2020-04-27 11:57:20.725 19669-19774/com.android.example.camera2.video D/CameraFragment: Recording started

owahltinez, 2020-04-27:
Thanks for reporting! It looks like you are just tapping the button, the app is designed for users to press and hold to record video. Until we fix it, as a workaround, you can simply increase `MIN_REQUIRED_RECORDING_TIME_MILLIS` to something like `5000L`.

Currently, we impose a minimum 1 second until the codec is released. Instead, we should watch for the codec state to ensure that there's no crash.

BrunoAndroidDev, 2020-04-27:
Oh, OK ! It's a strange behavior but OK.
Is it normal to have a flickering in the preview ?

owahltinez, 2020-04-27:
Flickering is how this app signals users that the recording is taking place, I have seen that in a couple of other apps and it seemed like an easy thing to implement.

BrunoAndroidDev, 2020-04-27:
It's not a bug, it's a feature ;)

closed 2020-04-27 by BrunoAndroidDev