android/camera-samples #218: CameraXBasic example - image previews not working with beta03 on SM-G930F

ruedijc, 2020-04-28:
Using android 7.0 on SM-G930F, portrait orientation shows camera preview in top half of screen only.
Landscape orientation shows black camera preview.
Error on orientation change is :
`2020-04-28 12:20:42.332 20833-20833/com.android.example.cameraxbasic E/ViewRootImpl: sendUserActionEvent() mView == null`

Images capture works successfully.
Camerax Alpha versions did work previously.

mgarciaguerrero, 2020-04-30:
I had the same issue @ruedijc. Check it here: https://github.com/android/camera-samples/issues/212#issue-606414559

owahltinez, 2020-05-04:
Thanks @mgarciaguerrero for chiming in and @ruedijc for reporting this. Indeed, it sounds like the same issue so I'll close it as a dupe.

closed 2020-05-04 by owahltinez