android/camera-samples #222: CameraX crashes when running in background

Michael14K, 2020-05-04:
Hi, 
after running CameraXBasic on my phone the app crashes nearly always when I move the app into background (home button or "other apps" button). 
Please see attachments for further information.
My device is a Motorola G3 running on 6.0.1 and no alterations to the code where made prior.
I am fairly new in android development so my apologize for not being more specific. Hope I could help.

Kind regards,
Michael

[image]
[Camera_x.pdf](https://github.com/android/camera-samples/files/4577060/Camera_x.pdf)
[bugreport-2020-05-04-20-38-06.zip](https://github.com/android/camera-samples/files/4577061/bugreport-2020-05-04-20-38-06.zip)

Michael14K, 2020-05-11:
I also noticed that the preview of the CameraXBasic is quite slow and lagging.

owahltinez, 2020-05-11:
Thanks for creating this issue and sending the bug report. The team is looking into it :-)

Sehyris, 2020-05-14:
I'm also facing the same issue. 
Do you have any updates?

owahltinez, 2020-05-15:
This issue has been acknowledge and will be fixed in a future release of the library. A workaround is for the app to use their own executor and not shut that executor down until all images have been returned (or accounted for).

owahltinez, 2020-06-09:
This issue should now be fixed after updating to `beta04`, please reopen if that's not the case

closed 2020-06-09 by owahltinez