android/camera-samples #231: Camera2 - Capture a still image with a different format than the Preview

BrunoAndroidDev, 2020-05-11:
Hello,

I'm developing a **Face Detection** feature with **Camera2** and **MLKit** and I used the [Camera2Basic sample](https://github.com/android/camera-samples/tree/master/Camera2Basic) to display the live preview. Related to MLKit performance tips, the preview uses the `ImageFormat.YUV_420_888` format in 480x360 pixels.

I've succeeded in capture a still image is the same format and resolution as the Preview but I want to capture a still image:
- a full resolution image or, as least, bigger than 480x360
- in JPEG format to be able to save it

How can I do that ?
[I've posted my attempts in StackOverflow](https://stackoverflow.com/questions/61410832/need-to-capture-a-still-image-during-face-detection-with-mlkit-and-camera2) with more context, you can refer to it

Thanks for your feedbacks

owahltinez, 2020-05-11:
The issue tracker is being used for problems with the unmodified samples. If you have questions about usage, please use tag `android-camerax` in StackOverflow or start a thread in the [developer forums](https://g.co/camerax/developers)

owahltinez, 2020-05-11:
Sorry, I misread your question and I just noticed it's not about CameraX. Please refer to the Camera2Basic sample, it does not capture the image in the same resolution as the preview. The size of the output depends on the size of the target surface being attached to the camera session.

closed 2020-05-11 by owahltinez