android/camera-samples #235: Reuse MediaRecorder when recording several videos

chaokang, 2020-05-13:
Hello,

According to the [documentation about `MediaRecorder`](https://developer.android.com/reference/kotlin/android/media/MediaRecorder#stop):
> Once recording is stopped, you will have to configure it again as if it has just been constructed.

This does not seem to be the case in [Camera2Video](https://github.com/android/camera-samples/blob/master/Camera2Video/app/src/main/java/com/example/android/camera2/video/fragments/CameraFragment.kt). As far as I understand, the `MediaRecorder` object is _not_ configured again each time after `stop()` and before `start()`.
Am I missing something?

Thanks!

Best regards,
Chao

owahltinez, 2020-05-15:
Thanks for flagging this. What I think you are missing is that the recorder is being configured *every time* before recording begins. Here's where that happens: https://github.com/android/camera-samples/blob/9c9c581d43d6e1239d80a2f449ef678d5042b2c5/Camera2Video/app/src/main/java/com/example/android/camera2/video/fragments/CameraFragment.kt#L270

Since the app exits the activity after recording finishes, the recorder is disposed and never reused because the recorder will be recreated when a new activity is created.

closed 2020-05-15 by owahltinez

dima-ko, 2021-03-29:
@owahltinez do you have an example where recorder is reused without closing activity? Preferable even without closing preview session?