android/camera-samples #237: CameraXBasic show black screen on preview

sam5727, 2020-05-14:
Hi, I found a issue on camera sample CameraXBasic
If I rapidly keep switching front and rear lens many times, it'll show black screen on preview.
And the capture button won't work either.
I've tested on my phone (pixel 4) and emulator.

thanks!!

owahltinez, 2020-05-15:
Thanks for submitting this. I'm unable to reproduce the issue. Are you seeing some kind of error in the logcat? Please attach a [bug report](https://developer.android.com/studio/debug/bug-report) if possible.

samhung556, 2020-05-18:
The way I reproduce the bug is keep clicking camera_switch_button rapidly for like 10 seconds.
Here is the error I got, and I try to capture image on the black screen.
```
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:42.151 26491-26534/com.android.example.cameraxbasic E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
2020-05-18 09:39:42.158 26491-26534/com.android.example.cameraxbasic E/CaptureSession: CameraCaptureSession.onConfiguredFailed() RELEASING
2020-05-18 09:39:49.554 26491-26491/com.android.example.cameraxbasic E/CameraXBasic: Photo capture failed: Camera is closed.
    androidx.camera.core.ImageCaptureException: Camera is closed.
        at androidx.camera.core.ImageCapture$ImageCaptureRequest.lambda$notifyCallbackError$1$ImageCapture$ImageCaptureRequest(ImageCapture.java:1963)
        at androidx.camera.core.-$$Lambda$ImageCapture$ImageCaptureRequest$1G7WSvt8TANxhZtOyewefm68pg4.run(Unknown Source:8)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
     Caused by: androidx.camera.core.CameraClosedException: Camera is closed.
        at androidx.camera.core.ImageCapture.abortImageCaptureRequests(ImageCapture.java:703)
        at androidx.camera.core.ImageCapture.onStateOffline(ImageCapture.java:699)
        at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$notifyStateOfflineToUseCases$12(Camera2CameraImpl.java:689)
        at androidx.camera.camera2.internal.-$$Lambda$Camera2CameraImpl$hp2EIAmfp4eGOX-qpuivXiBeeMA.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.app.ActivityThread.main(ActivityThread.java:7356) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930) 
2020-05-18 09:39:59.856 26491-26506/com.android.example.cameraxbasic E/bstractResolvableFuture: RuntimeException while executing runnable CallbackListener,androidx.camera.core.ImageCapture$4@62347c4 with executor java.util.concurrent.ThreadPoolExecutor@27b76ad[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 3]
    java.util.concurrent.RejectedExecutionException: Task CallbackListener,androidx.camera.core.ImageCapture$4@62347c4 rejected from java.util.concurrent.ThreadPoolExecutor@27b76ad[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 3]
        at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2086)
        at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:848)
        at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1394)
        at androidx.concurrent.futures.AbstractResolvableFuture.executeListener(AbstractResolvableFuture.java:1053)
        at androidx.concurrent.futures.AbstractResolvableFuture.complete(AbstractResolvableFuture.java:902)
[... cut ...]
        at androidx.concurrent.futures.AbstractResolvableFuture.setException(AbstractResolvableFuture.java:712)
        at androidx.concurrent.futures.CallbackToFutureAdapter$SafeFuture.setException(CallbackToFutureAdapter.java:184)
        at androidx.concurrent.futures.CallbackToFutureAdapter$Completer.finalize(CallbackToFutureAdapter.java:337)
        at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:289)
        at java.lang.Daemons$FinalizerDaemon.runInternal(Daemons.java:276)
        at java.lang.Daemons$Daemon.run(Daemons.java:137)
        at java.lang.Thread.run(Thread.java:919)
```

sam5727, 2020-05-18:
The way I reproduce the bug is keep clicking camera_switch_button rapidly for like 10 seconds.
Here is the error I got, and I try to capture image on the black screen.
```
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:25.618 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:29.061 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case online
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:32.678 26491-26534/com.android.example.cameraxbasic E/Camera: Failed to set already detached use case active
2020-05-18 09:39:42.151 26491-26534/com.android.example.cameraxbasic E/CameraCaptureSession: Session 0: Failed to create capture session; configuration failed
2020-05-18 09:39:42.158 26491-26534/com.android.example.cameraxbasic E/CaptureSession: CameraCaptureSession.onConfiguredFailed() RELEASING
2020-05-18 09:39:49.554 26491-26491/com.android.example.cameraxbasic E/CameraXBasic: Photo capture failed: Camera is closed.
    androidx.camera.core.ImageCaptureException: Camera is closed.
        at androidx.camera.core.ImageCapture$ImageCaptureRequest.lambda$notifyCallbackError$1$ImageCapture$ImageCaptureRequest(ImageCapture.java:1963)
        at androidx.camera.core.-$$Lambda$ImageCapture$ImageCaptureRequest$1G7WSvt8TANxhZtOyewefm68pg4.run(Unknown Source:8)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
     Caused by: androidx.camera.core.CameraClosedException: Camera is closed.
        at androidx.camera.core.ImageCapture.abortImageCaptureRequests(ImageCapture.java:703)
        at androidx.camera.core.ImageCapture.onStateOffline(ImageCapture.java:699)
        at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$notifyStateOfflineToUseCases$12(Camera2CameraImpl.java:689)
        at androidx.camera.camera2.internal.-$$Lambda$Camera2CameraImpl$hp2EIAmfp4eGOX-qpuivXiBeeMA.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:214) 
        at android.app.ActivityThread.main(ActivityThread.java:7356) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930) 
2020-05-18 09:39:59.856 26491-26506/com.android.example.cameraxbasic E/bstractResolvableFuture: RuntimeException while executing runnable CallbackListener,androidx.camera.core.ImageCapture$4@62347c4 with executor java.util.concurrent.ThreadPoolExecutor@27b76ad[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 3]
    java.util.concurrent.RejectedExecutionException: Task CallbackListener,androidx.camera.core.ImageCapture$4@62347c4 rejected from java.util.concurrent.ThreadPoolExecutor@27b76ad[Terminated, pool size = 0, active threads = 0, queued tasks = 0, completed tasks = 3]
        at java.util.concurrent.ThreadPoolExecutor$AbortPolicy.rejectedExecution(ThreadPoolExecutor.java:2086)
        at java.util.concurrent.ThreadPoolExecutor.reject(ThreadPoolExecutor.java:848)
        at java.util.concurrent.ThreadPoolExecutor.execute(ThreadPoolExecutor.java:1394)
        at androidx.concurrent.futures.AbstractResolvableFuture.executeListener(AbstractResolvableFuture.java:1053)
        at androidx.concurrent.futures.AbstractResolvableFuture.complete(AbstractResolvableFuture.java:902)
[... cut ...]
        at androidx.concurrent.futures.AbstractResolvableFuture.setException(AbstractResolvableFuture.java:712)
        at androidx.concurrent.futures.CallbackToFutureAdapter$SafeFuture.setException(CallbackToFutureAdapter.java:184)
        at androidx.concurrent.futures.CallbackToFutureAdapter$Completer.finalize(CallbackToFutureAdapter.java:337)
        at java.lang.Daemons$FinalizerDaemon.doFinalize(Daemons.java:289)
        at java.lang.Daemons$FinalizerDaemon.runInternal(Daemons.java:276)
        at java.lang.Daemons$Daemon.run(Daemons.java:137)
        at java.lang.Thread.run(Thread.java:919)
```

owahltinez, 2020-06-09:
@sam5727 can you please verify if this issue persists after the update to `beta04`?

sam5727, 2020-06-11:
Hi guys,
I add the request to fix #253 
Please help to check it!

Speedrockracer, 2020-06-17:
Having the same issue here. Mainly happens when I rapidly switch cameras and then close and re-open the app while it's still switching.
Same observations as @madisp

santiagogalindo87, 2020-06-19:
We have had a very similar problem.

In our case we do not reproduce this problem switching the cameras. We reproduce the problem when close and re-open the app

We are testing on a Samsung Galaxy J5 with Android 7.0 (Samsung Experience 8.1)

We leave you the log reproduced in this demo

Thanks in advance.

2020-06-18 20:27:34.379 5324-5324/com.android.example.cameraxbasic D/ViewRootImpl@c2ffb0a[MainActivity]: ViewPostImeInputStage processPointer 0
2020-06-18 20:27:34.487 5324-5324/com.android.example.cameraxbasic D/ViewRootImpl@c2ffb0a[MainActivity]: ViewPostImeInputStage processPointer 1
2020-06-18 20:27:34.490 5324-5324/com.android.example.cameraxbasic D/ImageCapture: Size of image capture request queue: 0
2020-06-18 20:27:35.493 5324-5634/com.android.example.cameraxbasic D/Camera: issue capture request for camera 1
2020-06-18 20:27:35.493 5324-5634/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: OPENING --> REOPENING
2020-06-18 20:27:35.493 5324-5634/com.android.example.cameraxbasic D/Camera: Resetting Capture Session
2020-06-18 20:27:35.493 5324-5634/com.android.example.cameraxbasic D/Camera: releasing session in state REOPENING
2020-06-18 20:27:35.494 5324-5634/com.android.example.cameraxbasic D/Camera: CameraDevice.onClosed(): 1
2020-06-18 20:27:35.495 5324-5634/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: REOPENING --> OPENING
2020-06-18 20:27:35.495 5324-5634/com.android.example.cameraxbasic D/Camera: Opening camera: 1
2020-06-18 20:27:35.495 5324-6244/com.android.example.cameraxbasic E/ImageCapture: takePictureInternal onFailure
    androidx.camera.core.CameraClosedException: Capture request is cancelled because camera is closed
        at androidx.camera.core.ImageCapture$8.onCaptureCancelled(ImageCapture.java:1247)
        at androidx.camera.camera2.internal.CaptureSession.cancelIssuedCaptureRequests(CaptureSession.java:744)
        at androidx.camera.camera2.internal.Camera2CameraImpl.closeCamera(Camera2CameraImpl.java:405)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.reopenCameraAfterError(Camera2CameraImpl.java:1316)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.handleErrorOnOpen(Camera2CameraImpl.java:1290)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onError(Camera2CameraImpl.java:1268)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onError(CameraDeviceStateCallbacks.java:119)
        at androidx.camera.camera2.internal.compat.CameraDeviceCompat$StateCallbackExecutorWrapper$3.run(CameraDeviceCompat.java:177)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2020-06-18 20:27:35.495 5324-5634/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-5a7a631d-17de-417e-8cda-3ba48cb7be6f, androidx.camera.core.ImageCapture-bec9a8da-2f66-46ac-a6fa-8ff4dec9054c, androidx.camera.core.ImageAnalysis-42c9bf20-872e-4827-b00f-a5a5f2b1e7e4] for camera: 1
2020-06-18 20:27:35.496 5324-5324/com.android.example.cameraxbasic E/CameraXBasic: Photo capture failed: Capture request is cancelled because camera is closed
    androidx.camera.core.ImageCaptureException: Capture request is cancelled because camera is closed
        at androidx.camera.core.ImageCapture$ImageCaptureRequest.lambda$notifyCallbackError$1$ImageCapture$ImageCaptureRequest(ImageCapture.java:1963)
        at androidx.camera.core.-$$Lambda$ImageCapture$ImageCaptureRequest$1G7WSvt8TANxhZtOyewefm68pg4.run(lambda)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6776)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1496)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1386)
     Caused by: androidx.camera.core.CameraClosedException: Capture request is cancelled because camera is closed
        at androidx.camera.core.ImageCapture$8.onCaptureCancelled(ImageCapture.java:1247)
        at androidx.camera.camera2.internal.CaptureSession.cancelIssuedCaptureRequests(CaptureSession.java:744)
        at androidx.camera.camera2.internal.Camera2CameraImpl.closeCamera(Camera2CameraImpl.java:405)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.reopenCameraAfterError(Camera2CameraImpl.java:1316)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.handleErrorOnOpen(Camera2CameraImpl.java:1290)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onError(Camera2CameraImpl.java:1268)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onError(CameraDeviceStateCallbacks.java:119)
        at androidx.camera.camera2.internal.compat.CameraDeviceCompat$StateCallbackExecutorWrapper$3.run(CameraDeviceCompat.java:177)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
[... cut ...]
2020-06-18 20:27:41.508 5324-5634/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: REOPENING --> OPENING
2020-06-18 20:27:41.509 5324-5634/com.android.example.cameraxbasic D/Camera: Opening camera: 1
2020-06-18 20:27:41.510 5324-5634/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.Preview-5a7a631d-17de-417e-8cda-3ba48cb7be6f, androidx.camera.core.ImageCapture-bec9a8da-2f66-46ac-a6fa-8ff4dec9054c, androidx.camera.core.ImageAnalysis-42c9bf20-872e-4827-b00f-a5a5f2b1e7e4] for camera: 1

owahltinez, 2020-06-24:
This will be fixed with the next version of the library, coming out this week. I'll close this once the sample is updated.

sam5727, 2020-08-25:
Hi, the version has been updated to beta08. I've tested it and it works fine!!
Check the pr #284 that I recently created.

closed 2020-09-10 by owahltinez

andrew8er, 2021-03-11:
I checked the preview timeout issue: I suspect it is a problem with either CameraX or some deeper level, like the HAL or camera driver. The fragment and the view model are behaving correctly, only the preview stays black. I also tried the updated version `1.1.0-alpha02` of CameraX to no avail. Also Google does not help either.

The situation can be remedied with navigating back and there again… really not the best solution. Still, I do not think, this is a very serious problem and would leave it as it is for now. Fixing this could take ages, and if we are lucky, this will fix itself with some update of CameraX.

PR #284 (2020-09-10): Update CameraX to beta08
Fix #237 
- Update core to beta08 and view to alpha15