android/camera-samples #241: samples in java

rruffer, 2020-05-19:
Please. I know that nowadays they are using the kotlin a lot. But java is still being used a lot. And converting a kotlin project to java is very difficult. Could you make samples in java? would help me a lot.

owahltinez, 2020-05-20:
Unfortunately, Android is Kotlin-first; we focus on creating and maintaining samples in Kotlin. We evaluate whether a sample needs a Java version on a case-by-case basis, and if there's enough developer interest we will port this sample to Java.

closed 2020-05-20 by owahltinez

sliontc, 2020-05-28:
I'm also expecting the Java version.

LiorA1, 2020-06-02:
old versions exists in the archive..

topherbuckley, 2020-07-29:
How is developer interest gauged? Is there somewhere we can bump/vote/push for this?