android/camera-samples #247: Slow Motion fails due to invalid preview size selection.

perracolabs, 2020-05-25:
The slow-motion sample fails to open the camera. Can't pass the selection screen. Seems that the sample may be selecting an invalid preview size, in this case 1920x1080, although slow-motion is limited at 1280x720 in my test device:

In the selection screen there is only 1 entry showing the correct specs: **Back(0) 1280x720 120FPS**
After pressing it, the app crashes with the next stack-trace. 

```
	2020-05-25 11:36:04.151 2347-2347/com.android.example.camera2.slowmo E/AndroidRuntime: FATAL EXCEPTION: main
Process: com.android.example.camera2.slowmo, PID: 2347
java.lang.IllegalArgumentException: Surface size 1920x1080 is not part of the high speed supported size list [1280x720]
	at android.hardware.camera2.utils.SurfaceUtils.checkConstrainedHighSpeedSurfaces(SurfaceUtils.java:180)
	at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSessionInternal(CameraDeviceImpl.java:689)
	at android.hardware.camera2.impl.CameraDeviceImpl.createConstrainedHighSpeedCaptureSession(CameraDeviceImpl.java:597)
	at com.example.android.camera2.slowmo.fragments.CameraFragment.createCaptureSession(CameraFragment.kt:387)
	at com.example.android.camera2.slowmo.fragments.CameraFragment$initializeCamera$1.invokeSuspend(CameraFragment.kt:262)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
	at android.os.Handler.handleCallback(Handler.java:907)
	at android.os.Handler.dispatchMessage(Handler.java:105)
	at android.os.Looper.loop(Looper.java:216)
	at android.app.ActivityThread.main(ActivityThread.java:7625)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
2020-05-25 11:36:04.172 2347-2347/com.android.example.camera2.slowmo I/Process: Sending signal. PID: 2347 SIG: 9
```

owahltinez, 2020-05-26:
Something strange is going on. What specific device and model (please include Android version) are you seeing this on?

perracolabs, 2020-05-27:
Device details: Huawei P20 Lite / Android Pie

I have been debugging the sample app, and I believe that the logic used in getPreviewOutputSize is not fully correct. 
The preview size should be resolved in relation to the sizes returned by StreamConfigurationMap.**getHighSpeedVideoSizes**().

The current implementation does not filter out the aspect ratios and sizes outside the ranges that getHighSpeedVideoSizes can return. So is returning a preview of 1920x1080, although this device supports only 1 slow motion size at 1280x720 (120fps)

owahltinez, 2020-05-28:
I see, thanks for the pointer. A potential solution in this case then would be to ensure that `getPreviewOutputSize` only outputs sizes that are compatible with `getHighSpeedVideoSizes()`.

Note that it will not always be the same as the chosen size for the high speed video capture, since there is no guarantee that sizes bigger than 1080P are supported for preview when used simultaneously during video recording.

perracolabs, 2020-05-28:
A point to have into consideration is that **createHighSpeedRequestList**  will always fail if it receives any surface that doesn't have a size which isn't in the list from **getHighSpeedVideoSizes**. So, it should be assumed that the sizes from getHighSpeedVideoSizes are always the ground truth.

You may check this logic in the method "**checkConstrainedHighSpeedSurfaces**" from **android.hardware.camera2.utils.SurfaceUtils**:

[https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/hardware/camera2/utils/SurfaceUtils.java](https://android.googlesource.com/platform/frameworks/base/+/master/core/java/android/hardware/camera2/utils/SurfaceUtils.java)

owahltinez, 2020-06-09:
Thanks for the pointer. This issue should now be fixed, please reopen if you are still seeing a crash.

closed 2020-06-09 by owahltinez