android/camera-samples #250: NVIDIA Shield Tablet K1 - Invalid surface (black screen)

bediaz, 2020-05-29:
Please follow the following instructions before filing a bug:

Execution:
Deploy CameraXBasic (Kotlin) to NVDIA Shield Tablet K1

Expected Result:
Preview will appear

Actual Result:
Black screen where camera preview should be.

Project: CameraXBasic (Kotlin)
Device Model: SHIELD Tablet K1
Build: Android 7.0
Logcat: 
`2020-05-29 13:12:02.564 5135-5164/com.android.example.cameraxbasic D/Camera: Use cases [ImageCapture:androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, Preview:androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99, ImageAnalysis:androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae] now ONLINE for camera 0
2020-05-29 13:12:02.565 5135-5164/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae, androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99] for camera: 0
2020-05-29 13:12:02.565 5135-5164/com.android.example.cameraxbasic D/Camera: Resetting Capture Session
2020-05-29 13:12:02.565 5135-5164/com.android.example.cameraxbasic D/Camera: releasing session in state INITIALIZED
2020-05-29 13:12:02.565 5135-5164/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: INITIALIZED --> OPENING
2020-05-29 13:12:02.565 5135-5164/com.android.example.cameraxbasic D/Camera: Opening camera: 0
2020-05-29 13:12:02.566 5135-5164/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae, androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99] for camera: 0
2020-05-29 13:12:02.613 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface created.
2020-05-29 13:12:02.613 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface changed. Size: 1920x1080
2020-05-29 13:12:02.893 5135-5164/com.android.example.cameraxbasic D/Camera: CameraDevice.onOpened(): 0
2020-05-29 13:12:02.894 5135-5164/com.android.example.cameraxbasic D/Camera: Transitioning camera internal state: OPENING --> OPENED
2020-05-29 13:12:02.895 5135-5164/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae, androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99] for camera: 0
2020-05-29 13:12:02.908 5135-5164/com.android.example.cameraxbasic D/CaptureSession: Some surfaces were closed.
2020-05-29 13:12:02.928 5135-5164/com.android.example.cameraxbasic D/Camera: Posting surface closed
    java.lang.Throwable
        at androidx.camera.camera2.internal.Camera2CameraImpl.postSurfaceClosedError(Camera2CameraImpl.java:920)
        at androidx.camera.camera2.internal.Camera2CameraImpl$3.onFailure(Camera2CameraImpl.java:861)
        at androidx.camera.core.impl.utils.futures.Futures$CallbackListener.run(Futures.java:338)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.os.HandlerThread.run(HandlerThread.java:61)
2020-05-29 13:12:02.929 5135-5164/com.android.example.cameraxbasic D/Camera: Use case Preview:androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99 RESET for camera 0
2020-05-29 13:12:02.929 5135-5164/com.android.example.cameraxbasic D/Camera: Resetting Capture Session
2020-05-29 13:12:02.929 5135-5164/com.android.example.cameraxbasic D/Camera: releasing session in state OPENED
2020-05-29 13:12:02.930 5135-5164/com.android.example.cameraxbasic D/UseCaseAttachState: Active and online use case: [androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae, androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99] for camera: 0
2020-05-29 13:12:02.930 5135-5164/com.android.example.cameraxbasic D/UseCaseAttachState: All use case: [androidx.camera.core.ImageCapture-9c8b90f0-0da1-4a3a-a78f-f22e474e52ee, androidx.camera.core.ImageAnalysis-87f4fe85-8d29-499b-a289-a526f535fdae, androidx.camera.core.Preview-1920c657-2360-4c7e-b056-ffe3c3114d99] for camera: 0
2020-05-29 13:12:02.931 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface destroyed.
2020-05-29 13:12:02.959 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface created.
2020-05-29 13:12:02.959 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface changed. Size: 1920x1080
2020-05-29 13:12:02.963 5135-5135/com.android.example.cameraxbasic D/SurfaceViewImpl: Surface set on Preview.
2020-05-29 13:12:02.965 5135-5164/com.android.example.cameraxbasic D/CaptureSession: Opening capture session.
2020-05-29 13:12:03.011 5135-5164/com.android.example.cameraxbasic D/CaptureSession: Attempting to send capture request onConfigured
2020-05-29 13:12:03.011 5135-5164/com.android.example.cameraxbasic D/CaptureSession: Issuing request for session.
2020-05-29 13:12:03.014 5135-5164/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onConfigured() mState=OPENED
2020-05-29 13:12:03.014 5135-5164/com.android.example.cameraxbasic D/CaptureSession: CameraCaptureSession.onReady() OPENED
2020-05-29 13:12:03.326 5135-5231/com.android.example.cameraxbasic D/CameraXBasic: Average luminosity: 40.541889105902776
2020-05-29 13:12:03.444 5135-5231/com.android.example.cameraxbasic D/CameraXBasic: Average luminosity: 41.88744357638889
2020-05-29 13:12:03.573 5135-5231/com.android.example.cameraxbasic D/CameraXBasic: Average luminosity: 22.45810980902778
2020-05-29 13:12:03.680 5135-5231/com.android.example.cameraxbasic D/CameraXBasic: Average luminosity: 25.406436631944445`

Video:
[image]

owahltinez, 2020-06-09:
@bediaz can you please try again now that the library has been updated to `beta04`?

bediaz, 2020-06-15:
The beta04 library resolves the issue with the black preview! However, the preview is now stretched when in portrait. I will look into this and create a separate issue.

closed 2020-06-15 by bediaz