android/camera-samples #254: Camera2Video crashes when tap capture button

ianyhlee, 2020-06-16:
Reproduce steps:
1. Launch camera2video sample app
2. Select "Back (0) 1280x720 30 FPS" (basically any resolution/fps combination has the same result)
3. Tap record circle button -> app crashes

Crash log:
`2020-06-16 12:16:21.228 16140-16140/? I/e.camera2.vide: Late-enabling -Xcheck:jni
2020-06-16 12:16:21.259 16140-16140/? E/e.camera2.vide: Unknown bits set in runtime_flags: 0x8000
2020-06-16 12:16:21.548 16140-16140/? I/Perf: Connecting to perf service.
2020-06-16 12:16:21.675 16140-16140/? W/e.camera2.vide: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (greylist, reflection, allowed)
2020-06-16 12:16:21.676 16140-16140/? W/e.camera2.vide: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (greylist, reflection, allowed)
2020-06-16 12:16:21.748 16140-16140/? I/CameraManagerGlobal: Connecting to camera service
2020-06-16 12:16:21.750 16140-16140/? E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:21.751 16140-16140/? I/chatty: uid=10227(com.android.example.camera2.video) identical 1 line
2020-06-16 12:16:21.751 16140-16140/? E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:21.751 16140-16160/? E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:21.854 16140-16214/? I/AdrenoGLES: QUALCOMM build                   : a1a50b4, Ia37726fadd
    Build Date                       : 05/06/20
    OpenGL ES Shader Compiler Version: EV031.27.05.08
    Local Branch                     : mybranchea8fba03-080e-f1c6-8ca7-08f1eafda2d0
    Remote Branch                    : quic/gfx-adreno.lnx.1.0.r108-rel
    Remote Branch                    : NONE
    Reconstruct Branch               : NOTHING
2020-06-16 12:16:21.854 16140-16214/? I/AdrenoGLES: Build Config                     : S L 8.0.12 AArch64
2020-06-16 12:16:21.858 16140-16214/? I/AdrenoGLES: PFP: 0x005ff112, ME: 0x005ff066
2020-06-16 12:16:21.923 16140-16214/? W/Gralloc3: mapper 3.x is not supported
2020-06-16 12:16:21.916 16140-16140/? W/RenderThread: type=1400 audit(0.0:6966): avc: denied { search } for name="/" dev="mmcblk0p37" ino=2 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:factory_file:s0 tclass=dir permissive=0
2020-06-16 12:16:26.541 16140-16140/com.android.example.camera2.video D/CameraFragment: View finder size: 2560 x 1600
2020-06-16 12:16:26.541 16140-16140/com.android.example.camera2.video D/CameraFragment: Selected preview size: 1920x1080
2020-06-16 12:16:26.560 16140-16140/com.android.example.camera2.video D/AutoFitSurfaceView: Measured dimensions set: 2844 x 1600
2020-06-16 12:16:26.560 16140-16140/com.android.example.camera2.video I/chatty: uid=10227(com.android.example.camera2.video) identical 1 line
2020-06-16 12:16:26.561 16140-16140/com.android.example.camera2.video D/AutoFitSurfaceView: Measured dimensions set: 2844 x 1600
2020-06-16 12:16:26.601 16140-16140/com.android.example.camera2.video E/libc: Access denied finding property "persist.vendor.camera.privapp.list"
2020-06-16 12:16:26.596 16140-16140/com.android.example.camera2.video W/e.camera2.video: type=1400 audit(0.0:6970): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2020-06-16 12:16:26.626 16140-16140/com.android.example.camera2.video W/Binder:16140_3: type=1400 audit(0.0:6971): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2020-06-16 12:16:26.632 16140-16160/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:26.636 16140-16140/com.android.example.camera2.video W/Binder:16140_3: type=1400 audit(0.0:6972): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2020-06-16 12:16:26.643 16140-16160/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:26.646 16140-16140/com.android.example.camera2.video W/Binder:16140_3: type=1400 audit(0.0:6974): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2020-06-16 12:16:26.651 16140-16160/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:26.651 16140-16158/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:26.715 16140-16140/com.android.example.camera2.video I/OMXClient: IOmx service obtained
2020-06-16 12:16:28.484 16140-16302/com.android.example.camera2.video D/CameraFragment: Recording stopped. Output file: /data/user/0/com.android.example.camera2.video/files/VID_2020_06_16_12_16_26_721.mp4
2020-06-16 12:16:28.484 16140-16302/com.android.example.camera2.video E/MediaRecorder: stop called in an invalid state: 4
2020-06-16 12:16:28.521 16140-16302/com.android.example.camera2.video E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-3
    Process: com.android.example.camera2.video, PID: 16140
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
2020-06-16 12:16:28.653 16140-16300/com.android.example.camera2.video D/CameraFragment: Recording started
2020-06-16 12:16:29.066 16140-16140/com.android.example.camera2.video W/Binder:16140_3: type=1400 audit(0.0:6978): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
2020-06-16 12:16:29.078 16140-16160/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:29.084 16140-16160/com.android.example.camera2.video E/libc: Access denied finding property "vendor.camera.aux.packagelist"
2020-06-16 12:16:29.076 16140-16140/com.android.example.camera2.video W/Binder:16140_3: type=1400 audit(0.0:6979): avc: denied { read } for name="u:object_r:persist_camera_prop:s0" dev="tmpfs" ino=13113 scontext=u:r:untrusted_app:s0:c227,c256,c512,c768 tcontext=u:object_r:persist_camera_prop:s0 tclass=file permissive=0
`

gonemad4u, 2020-06-25:
same, my error is also IllegalStateException

_2020-06-26 08:04:48.842 15842-16101/com.android.example.camera2.video E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-3
    Process: com.android.example.camera2.video, PID: 15842
    java.lang.IllegalStateException
        at android.media.MediaRecorder.native_stop(Native Method)
        at android.media.MediaRecorder.stop(MediaRecorder.java:1610)
        at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)_

owahltinez, 2020-06-26:
This is a UI issue, you need to press and hold to record. We will change the UI to make this more clear and prevent very short recordings, which is what is causing the crash. In the meantime, try again but long-pressing the button instead and this crash should go away.

KarimFikani, 2020-07-03:
@owahltinez the screen keeps on blinking when my finger is down on the button. Gives the impression that we are taking multiple photos instead of recording.

arstagaev, 2021-04-05:
i have same error, when i tap to circle button -> app crashing. when I hold circle button screen fast blinked and crashes. Phone: Samsung Galaxy M21