android/camera-samples #256: Camera2Basic - Preview is stretched

guillaume-tgl, 2020-06-17:
The Preview is stretched when running the Camera2Basic sample.

Device: Galaxy S9
Android version 10

[image]

[image]

ianyhlee, 2020-06-18:
Looks these samples are really buggy

guillaume-tgl, 2020-06-18:
Is it the samples or the Camera2 API?...

ianyhlee, 2020-06-18:
Not 100% sure, does below sample work correctly on your phone?
https://github.com/googlearchive/android-Camera2Basic
It should use camera2 API as well.

guillaume-tgl, 2020-06-18:
This sample works, yes. I suspect it is because it uses a TextureView while this sample uses a SurfaceView but I'm not sure.

ZTFtrue, 2020-07-15:
use this when ```surfaceCreated``` :
```
val la = Constraints.LayoutParams(1080, 1920);//  width , height , it neee you calculate
 viewFinder.setLayoutParams(la)
```
I guess becuse  ```AutoFitSurfaceView``` , ```onMeasure``` error.
I test  ```setAspectRatio  ```  , some times width and height  is reverse. if you  can know what is curretly, you can fix it.

ZTFtrue, 2020-07-15:
I only  need  portrait,
i can use this:
```
    override fun onMeasure(widthMeasureSpec: Int, heightMeasureSpec: Int) {
        super.onMeasure(widthMeasureSpec, heightMeasureSpec)
        val width = MeasureSpec.getSize(widthMeasureSpec)
        val height = MeasureSpec.getSize(heightMeasureSpec)
        if (aspectRatio == 0f) {
            setMeasuredDimension(width, height)
        } else {

            // Performs center-crop transformation of the camera frames
            val newWidth: Int
            val newHeight: Int

            if (width < height ) {
                newHeight = (width*aspectRatio).roundToInt()
                newWidth = width;
            } else {
                newWidth = (height*aspectRatio).roundToInt()
                newHeight = height
            }

            Log.d(TAG, "Measured dimensions set: $newWidth x $newHeight")
            setMeasuredDimension(newWidth, newHeight)
        }
    }
```

ilexbor, 2020-07-17:
Redmi Note 8 Pro preview is stretched

snti, 2020-07-31:
The Preview is stretched too in my Pixel 3a XL.

owahltinez, 2020-08-20:
It should now be fixed thanks to #281

closed 2020-08-20 by owahltinez