android/camera-samples #259: App crashes when selecting a high resolution and recording from list in SelectorFragment

BitfulThinking, 2020-06-18:
Here are the error logs:

`E/MediaRecorder: stop failed: -1007`
`E/CameraFragment: java.lang.RuntimeException: stop failed.`

The logs aren't helpful but I think I know what's the issue. 
Try building and installing the Camera2Video project on a real android device. (I used a Nexus 6P 8.0)

Select a high resolution from the list of supported resolutions in SelectorFragment.

After you attempt recording from CameraFragment and finish recording the app will crash. I think it's because the high resolution that gets selected and then passed into `setVideoSize(args.width, args.height)` in the method `createRecorder()` is the culprit. 

IIRC MediaRecorder cannot handle high-resolution video, so sizes larger than 1080p should be filtered out in your `SelectorFragment#enumerateVideoCameras` method. 

I can submit a patch for this on the repo but it really is design choice how you want to handle it. For example, iterating through all the supported Size objects and only selecting ones that have aspect ratio of 3x4 or 16x9 and are less than 1080p. 

Cheers!

BitfulThinking, 2020-06-19:
Here's a potential fix: 

```
private fun getBestVideoSize(sizes: Array<Size>) : Size? {
        return sizes.firstOrNull { size -> size.height == SIZE_1080P.size.height && size.width == SIZE_1080P.size.width }
            ?: sizes.firstOrNull { size -> size.width == size.height * 4 / 3 && size.width <= 1080 }
            ?: sizes.firstOrNull { size -> size.width == size.height * 16 / 9 && size.width <= 1080 }
            ?: sizes.firstOrNull()
    }

owahltinez, 2020-06-26:
Thanks for reporting. MediaRecorder does support resolutions higher than 1080p, so I don't think that's the issue and it must be something else. Does it work when you use a smaller resolution? Have you tried taking a longer video, e.g. 10 seconds or longer?

owahltinez, 2020-08-20:
Closing due to inactivity, please reopen if it's still an issue.

closed 2020-08-20 by owahltinez