android/camera-samples #267: Camera2Video app crashes when record button is tapped

BlazingSkylark, 2020-07-09:
**Log on IDE:**
E/MediaRecorder: stop called in an invalid state: 4

**Log on device:**
java.lang.IllegalStateException
	at android.media.MediaRecorder.stop(Native Method)
	at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)

**Device:** 
Redmi Note 8
Android 9

npb12, 2020-07-09:
Same issue here

owahltinez, 2020-07-10:
Thanks for reporting, this is a dupe of #254

closed 2020-07-10 by owahltinez