android/camera-samples #268: Camera2api had trouble taking photos continuously

VuManhHung97, 2020-07-09:
I have problems using camera2api from this model. I just removed the image viewer function.
I tested on a number of devices such as Nokia 6, Nokia 7.2, Nokia 8.1, Samsung j7 prime, and most likely caused the error.
Another small problem is that the preview image is stretched and the phone will get hot so fast
This is an error video : https://drive.google.com/file/d/1ZXLcedqqx_SKKxNrJfWjC4-P8cGPHtFb/view?usp=sharing

Ragefalcon, 2020-07-30:
I just dealt with that problem. I'm not very experienced in android and maybe the problem is somewhere even deeper, but I see this:
When you get an "image" in function `takePhoto()`, there is code:

```
...
       session.capture(captureRequest.build(), object : CameraCaptureSession.CaptureCallback() {`
...
               // Loop in the coroutine's context until an image with matching timestamp 
                // We need to launch the coroutine context again because the callback is done in
                //  the handler provided to the `capture` method, not in our coroutine context
                @Suppress("BlockingMethodInNonBlockingContext")
                lifecycleScope.launch(cont.context) {
                    while (true) {
...
                        // Build the result and resume progress
                        cont.resume(CombinedCaptureResult(
                                image, result, exifOrientation, imageReader.imageFormat))

                        // There is no need to break out of the loop, this coroutine will suspend
                    }
...
```

It is even noted here that the loop is written correctly, but it seems that it is because of this loop that the thread remains hanging and after accumulating 64 (you can even count flashes on the video) the threads in `Dispatchers.IO` stop running. Since it is 64 threads that are restricted in `Dispatchers.IO` by default. And if you add the key to exit the loop, everything starts to work steadily:
```
...
                var key = true               
                @Suppress("BlockingMethodInNonBlockingContext")
                lifecycleScope.launch(cont.context) {
                    while (key ) {
...
                        // Build the result and resume progress
                        key = false                        
                        cont.resume(CombinedCaptureResult(
                                image, result, exifOrientation, imageReader.imageFormat))

                        // There is no need to break out of the loop, this coroutine will suspend
                    }
...
```

I don't know if it's a realization problem or maybe it's coroutine problem, but it helped me.
I'm sorry if my English isn't very clear...)

owahltinez, 2020-08-20:
The image being stretched was fixed by #256. For the other issues that arise after you modify the sample, we would need a little more information to help you. I would recommend creating a question on StackOverflow, feel free to post the link here.

closed 2020-08-20 by owahltinez