android/camera-samples #269: CameraX could not take photo on API 30 Pixel 3XL simulator

kostya29, 2020-07-09:
CameraX could not take photo on API 30 Pixel 3XL simulator
There is such exception:

```
2020-07-09 17:48:18.497 9775-9838/com.android.example.cameraxbasic E/CameraCaptureSession: Session 0: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2163)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1253)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:578)
        at androidx.camera.camera2.internal.CaptureSession.closeCameraCaptureSession(CaptureSession.java:572)
        at androidx.camera.camera2.internal.CaptureSession.release(CaptureSession.java:520)
        at androidx.camera.camera2.internal.Camera2CameraImpl.releaseSession(Camera2CameraImpl.java:493)
        at androidx.camera.camera2.internal.Camera2CameraImpl.resetCaptureSession(Camera2CameraImpl.java:1045)
        at androidx.camera.camera2.internal.Camera2CameraImpl.closeCamera(Camera2CameraImpl.java:410)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.reopenCameraAfterError(Camera2CameraImpl.java:1460)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.handleErrorOnOpen(Camera2CameraImpl.java:1434)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onError(Camera2CameraImpl.java:1412)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onError(CameraDeviceStateCallbacks.java:119)
        at android.hardware.camera2.impl.CameraDeviceImpl.notifyError(CameraDeviceImpl.java:1582)
        at android.hardware.camera2.impl.CameraDeviceImpl.lambda$oDs27OTfKFfK18rUW2nQxxkPdV0(Unknown Source:0)
        at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$oDs27OTfKFfK18rUW2nQxxkPdV0.accept(Unknown Source:8)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:278)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:201)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:97)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$1.run(SequentialExecutor.java:110)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:230)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:172)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
```

narkusv, 2020-08-17:
Same here. It seems that only the front camera is affected though..

narkusv, 2020-08-19:
This seems to be a emulator issue https://issuetracker.google.com/issues/161034252

owahltinez, 2020-08-20:
Thanks for chasing that down, @narkusv. Since it's a library issue and we have a bug for it, I'll close this one.

closed 2020-08-20 by owahltinez