android/camera-samples #272: CameraX can't adapt redmi note2 (android 5.0.2)

zhouyueyuedsf, 2020-07-15:
#### Preview will be  stretched on redmi note2.
I config rightly params, and see expected params (suggestedResolutionsMap) by source code debug 。 but preview is still stretched。
#### sourceCode
```java
Map<UseCaseConfig<?>, Size> getSuggestedResolutions(
            List<SurfaceConfig> existingSurfaces, List<UseCaseConfig<?>> newUseCaseConfigs) {
        Map<UseCaseConfig<?>, Size> suggestedResolutionsMap = new HashMap<>();

        // Get the index order list by the use case priority for finding stream configuration
        List<Integer> useCasesPriorityOrder = getUseCasesPriorityOrder(newUseCaseConfigs);
        List<List<Size>> supportedOutputSizesList = new ArrayList<>();

        // Collect supported output sizes for all use cases
        for (Integer index : useCasesPriorityOrder) {
            List<Size> supportedOutputSizes =
                    getSupportedOutputSizes(newUseCaseConfigs.get(index));
            supportedOutputSizesList.add(supportedOutputSizes);
        }

        // Get all possible size arrangements
        List<List<Size>> allPossibleSizeArrangements =
                getAllPossibleSizeArrangements(supportedOutputSizesList);

        // Transform use cases to SurfaceConfig list and find the first (best) workable combination
        for (List<Size> possibleSizeList : allPossibleSizeArrangements) {
            // Attach SurfaceConfig of original use cases since it will impact the new use cases
            List<SurfaceConfig> surfaceConfigList = new ArrayList<>(existingSurfaces);

            // Attach SurfaceConfig of new use cases
            for (int i = 0; i < possibleSizeList.size(); i++) {
                Size size = possibleSizeList.get(i);
                UseCaseConfig<?> newUseCase =
                        newUseCaseConfigs.get(useCasesPriorityOrder.get(i));
                surfaceConfigList.add(transformSurfaceConfig(newUseCase.getInputFormat(), size));
            }

            // Check whether the SurfaceConfig combination can be supported
            if (checkSupported(surfaceConfigList)) {
                for (UseCaseConfig<?> useCaseConfig : newUseCaseConfigs) {
                    suggestedResolutionsMap.put(
                            useCaseConfig,
                            possibleSizeList.get(
                                    useCasesPriorityOrder.indexOf(
                                            newUseCaseConfigs.indexOf(useCaseConfig))));
                }
                break;
            }
        }
      
        return suggestedResolutionsMap;
    }
```
#### Directly Reason
> params `surfaces` in config is error. `surface[0]` is not Preview surface, but ImageCapture surface
@Override
    public void createCaptureSession(@NonNull SessionConfigurationCompat config)
            throws CameraAccessException {
        checkPreconditions(mCameraDevice, config);

        if (config.getInputConfiguration() != null) {
            throw new IllegalArgumentException("Reprocessing sessions not supported until API 23");
        }

        if (config.getSessionType() == SessionConfigurationCompat.SESSION_HIGH_SPEED) {
            throw new IllegalArgumentException(
                    "High speed capture sessions not supported until API 23");
        }

        // Wrap the executor in the callback
        CameraCaptureSession.StateCallback cb =
                new CameraCaptureSessionCompat.StateCallbackExecutorWrapper(
                        config.getExecutor(), config.getStateCallback());

        // Convert the OutputConfigurations to surfaces
        List<Surface> surfaces = unpackSurfaces(config.getOutputConfigurations());

        CameraDeviceCompatParamsApi21 params = (CameraDeviceCompatParamsApi21) mImplParams;
        createBaseCaptureSession(mCameraDevice, surfaces, cb, params.mCompatHandler);
    }

ilexbor, 2020-07-17:
Redmi Note 8 Pro preview is stretched

zhouyueyuedsf, 2020-07-17:
> Redmi Note 8 Pro preview is stretched
redmi note 8 is ok. i test some phones, and summary an excel
[image]

owahltinez, 2020-08-20:
It seems like an issue with the library. Can you please start a thread in the [developer forums](https://g.co/camerax/developers) and provide more details (like a screenshot) to the team?

closed 2020-08-20 by owahltinez