android/camera-samples #273: [CameraX-Camera2Api] App crashes on preview

hikkidev, 2020-07-20:
After opening the camera, a black preview is visible and the application is closed.

I get a lot of crashes from Firebase-Crashlytics
Android v: 7, 9, 10
Devices: Redmi Note 5 Pro, Galaxy S8, Nexus 4 (Emulator)

Library:
```groovy
/* CameraX */
val cameraxVersion = "1.0.0-beta04"
implementation("androidx.camera:camera-core:$cameraxVersion")
implementation("androidx.camera:camera-camera2:$cameraxVersion")
implementation("androidx.camera:camera-lifecycle:$cameraxVersion")
// CameraX View class
implementation("androidx.camera:camera-view:1.0.0-alpha13")
```

Error log:
```java
Fatal Exception: java.lang.NoSuchMethodError: No virtual method setUseCaseActive(Landroidx/camera/core/UseCase;)V in class Landroidx/camera/core/impl/UseCaseAttachState; or its super classes (declaration of 'androidx.camera.core.impl.UseCaseAttachState' appears in /data/app/com.myapp.some.domain-1/base.apk)
       at androidx.camera.camera2.internal.Camera2CameraImpl.lambda$onUseCaseActive$4$Camera2CameraImpl(Camera2CameraImpl.java:557)
       at androidx.camera.camera2.internal.-$$Lambda$Camera2CameraImpl$yjqZb_jyZ_DzssP2QQvZFrqJco4.run(lambda)
       at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:230)
       at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:172)
       at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1133)
       at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:607)
       at java.lang.Thread.run(Thread.java:761)
```

husaynhakeem, 2020-07-25:
This looks like it might be because you missed adding support for java 8 features in the project? Make sure you're adding this in your gradle file.

```
compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }
```

Btw, the CameraX versions you're using don't match. The `beta04` version of core/camera2/lifecycle should be used with the `alpha11` version of camera-view. Even better, you can update to the latest versions (`beta07` of camera2/lifecycle and `alpha14` of view).

hikkidev, 2020-07-25:
> Btw, the CameraX versions you're using don't match. The beta04 version of core/camera2/lifecycle should be used with the alpha11 version of camera-view. Even better, you can update to the latest versions (beta07 of camera2/lifecycle and alpha14 of view).
I suspect the problem was version mismatch. I have already updated the dependencies to the latest version while waiting for the results.

owahltinez, 2020-08-20:
From your comments I'm gathering that the issue is now resolved, please reopen if that's not the case.

closed 2020-08-20 by owahltinez