android/camera-samples #274: Orientation Issue on Pixel 2 API 26 Emulator

onurays, 2020-07-22:
Images are always rotated 90 degrees as below. Do you think this is a CameraX bug or because this sample (and Glide) is not respecting the Exif Info?

Camera feed:
[image]

Preview:
[image]

EdwardvanRaak, 2020-07-23:
Have you checked the rotation values in:

```
        imageCapture = ImageCapture.Builder()
            .setTargetRotation(rotation)
```

and

`imageCapture?.targetRotation = view.display.rotation` in the `displayListener`?

They should be 0 or `ROTATION_0`

owahltinez, 2020-11-08:
I'm unable to reproduce this issue, please reopen if it persists with the latest version of the emulator and Android Studio.

closed 2020-11-08 by owahltinez