android/camera-samples #277: Q: Yuv converter

gordinmitya, 2020-08-12:
Hello, thank you for your job!
I have a question related to camera2/camerax YUV format.
In documentation it is specified as YUV_420_888 generic standard, so in theory an image may came in NV21 (U and V are interleaved, pixelStride=2) or I420 (U and V are separated, pixelStride=1).
The question: is there actually devices which provide images in I420 format? Or we can always be sure that image is NV21?

Sub-question: why do you need to copy pixel-by-pixel when pixelStride == outputStride == 2 (NV21 case) [here](https://github.com/android/camera-samples/blob/master/CameraXTfLite/utils/src/main/java/com/example/android/camera/utils/YuvToRgbConverter.kt#L194) while you can just copy the whole `uvuvuvuv...` row?

gordinmitya, 2020-08-12:
even if assumption about NV21 is not true, you can avoid direct pixel-manipulations just by adjusting RenderScript Input type here:
```kotlin
val elemType = Type.Builder(rs, Element.YUV(rs)).setYuvFormat(ImageFormat.NV21).create()
```

```kotlin
val elemType = Type.Builder(rs, Element.YUV(rs)).setYuvFormat(ImageFormat.YUV_420_888).create()
```

owahltinez, 2020-08-20:
Thanks for the feedback @gordinmitya! I believe that almost all cameras will output NV21 format, but it's not a guarantee. Regarding your suggestions to optimize this, it's likely that the code can be improved significantly since it was only meant as a "proof of concept". If you wish to open a PR to include your suggestions (and make sure it works on at least a few devices) then we would be happy to include that.

closed 2020-10-01 by owahltinez