android/camera-samples #279: CameraAccessException in Samsung device Camera2Video

buntupana, 2020-08-14:
Experiencing an error in a Samsung Galaxy 7 Active

Log:
androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent
WindowCallbackWrapper.java, line 69
android.hardware.camera2.CameraAccessException: waitUntilIdle:1101: Camera 0: Error waiting to drain: Connection timed out 
android.hardware.camera2.CameraManager.throwAsPublicException CameraManager.java:707
android.hardware.camera2.impl.ICameraDeviceUserWrapper.waitUntilIdle ICameraDeviceUserWrapper.java:177
android.hardware.camera2.impl.CameraDeviceImpl.waitUntilIdle CameraDeviceImpl.java:1029
android.hardware.camera2.impl.CameraDeviceImpl.configureStreamsChecked CameraDeviceImpl.java:431
android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSessionInternal CameraDeviceImpl.java:632
android.hardware.camera2.impl.CameraDeviceImpl.createCaptureSession CameraDeviceImpl.java:501
com.snapcell.androidapp.presentation.record.VideoCameraImplNative.startPreview VideoCameraImplNative.kt:456
com.snapcell.androidapp.presentation.record.VideoCameraImplNative.stopInternalRecording VideoCameraImplNative.kt:302
com.snapcell.androidapp.presentation.record.VideoCameraImplNative.stopRecordingVideo VideoCameraImplNative.kt:285
com.snapcell.androidapp.presentation.record.RecordFragment.stopRecording RecordFragment.kt:272
com.snapcell.androidapp.presentation.record.RecordFragment.access$stopRecording RecordFragment.kt:40
com.snapcell.androidapp.presentation.record.RecordFragment$onCreateView$2.onTouch RecordFragment.kt:154
android.view.View.dispatchTouchEvent View.java:12539
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
android.view.ViewGroup.dispatchTransformedTouchEvent ViewGroup.java:3159
android.view.ViewGroup.dispatchTouchEvent ViewGroup.java:2844
com.android.internal.policy.DecorView.superDispatchTouchEvent DecorView.java:601
com.android.internal.policy.PhoneWindow.superDispatchTouchEvent PhoneWindow.java:1871
android.app.Activity.dispatchTouchEvent Activity.java:3384
androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent WindowCallbackWrapper.java:69
androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent WindowCallbackWrapper.java:69
com.android.internal.policy.DecorView.dispatchTouchEvent DecorView.java:563
android.view.View.dispatchPointerEvent View.java:12791
android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent ViewRootImpl.java:5670
android.view.ViewRootImpl$ViewPostImeInputStage.onProcess ViewRootImpl.java:5465
android.view.ViewRootImpl$InputStage.deliver ViewRootImpl.java:4958
android.view.ViewRootImpl$InputStage.onDeliverToNext ViewRootImpl.java:5011
android.view.ViewRootImpl$InputStage.forward ViewRootImpl.java:4977
android.view.ViewRootImpl$AsyncInputStage.forward ViewRootImpl.java:5114
android.view.ViewRootImpl$InputStage.apply ViewRootImpl.java:4985
android.view.ViewRootImpl$AsyncInputStage.apply ViewRootImpl.java:5171
android.view.ViewRootImpl$InputStage.deliver ViewRootImpl.java:4958
android.view.ViewRootImpl$InputStage.onDeliverToNext ViewRootImpl.java:5011
android.view.ViewRootImpl$InputStage.forward ViewRootImpl.java:4977
android.view.ViewRootImpl$InputStage.apply ViewRootImpl.java:4985
android.view.ViewRootImpl$InputStage.deliver ViewRootImpl.java:4958
android.view.ViewRootImpl.deliverInputEvent ViewRootImpl.java:7736
android.view.ViewRootImpl.doProcessInputEvents ViewRootImpl.java:7676
android.view.ViewRootImpl.enqueueInputEvent ViewRootImpl.java:7637
android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent ViewRootImpl.java:7847
android.view.InputEventReceiver.dispatchInputEvent InputEventReceiver.java:197
android.os.MessageQueue.nativePollOnce MessageQueue.java
android.os.MessageQueue.next MessageQueue.java:325
android.os.Looper.loop Looper.java:142
android.app.ActivityThread.main ActivityThread.java:6944
java.lang.reflect.Method.invoke Method.java
com.android.internal.os.Zygote$MethodAndArgsCaller.run Zygote.java:327
com.android.internal.os.ZygoteInit.main ZygoteInit.java:1374

owahltinez, 2020-08-20:
This exception typically happens as a result of the device being in a bad state. Does it persist after you reboot your device?

owahltinez, 2020-11-08:
Closing due to inactivity

closed 2020-11-08 by owahltinez