android/camera-samples #292: Camera2SlowMotion shows black screen for all resolutions in Mi Note 10 pro, Samsung s9+, 1+

milap612, 2020-08-27:
Preview displays black screen for all resolutions for high speed in both facing camera.
I have tested in these devices:-
Samsung s9+
One+
Mi Note 10 pro

All these device having android Q(10) version.

milapchokshi-simformsolutions, 2020-09-03:
@owahltinez please do the needful to get it done.

milap612, 2020-09-03:
@owahltinez please do the needful to get it done

owahltinez, 2020-09-14:
I am not able to reproduce this issue on any of my devices. Are you seeing anything in logcat?

milap612, 2020-09-14:
@owahltinez  below is the log for RedMi Note 9 pro back camera 120 FPS 1280*720 Resolution.

2020-09-14 12:41:41.397 786-786/? E/CamX: [ERROR][HAL    ] camxsession.cpp:4566 CheckValidInputRequest() ERROR Invalid metadata provided input 0x0 output 0x0
2020-09-14 12:41:41.397 786-786/? W/CamX: [ERROR][Always ] camxchicontext.cpp:5219 SubmitRequest(): Request batch index 0 is not valid.
2020-09-14 12:41:41.397 786-786/? E/CamX: [ERROR][CHI    ] camxchi.cpp:2430 ChiSubmitPipelineRequest() Submit request failed with error 4.
2020-09-14 12:41:41.397 786-786/? E/CHIUSECASE: [ERROR  ] chxadvancedcamerausecase.cpp:3701 ExecuteCaptureRequest() Submit request failure for session:0
2020-09-14 12:41:41.397 786-786/? E/CHIUSECASE: [ERROR  ] chxadvancedcamerausecase.cpp:3714 ExecuteCaptureRequest() Defer thread or submit request failure for session:0
2020-09-14 12:41:41.397 786-786/? E/CamX: [ERROR][Assert ] camxhaldevice.cpp:1315 ProcessCaptureRequest(): CamxResultSuccess == result
2020-09-14 12:41:41.397 786-786/? E/CamDevSession@3.4-impl: processOneCaptureRequest_3_4: HAL process_capture_request call failed!
2020-09-14 12:41:41.397 1076-12767/? E/Camera3-Device: Camera 0: sendRequestsBatch: RequestThread: Unable to submit capture request 0 to HAL device: Invalid argument (-22)
2020-09-14 12:41:41.398 10857-12764/com.android.example.camera2.slowmo E/CameraFragment: Camera 0 error: (4) Fatal (device)
    java.lang.RuntimeException: Camera 0 error: (4) Fatal (device)
        at com.example.android.camera2.slowmo.fragments.CameraFragment$openCamera$$inlined$suspendCancellableCoroutine$lambda$1.onError(CameraFragment.kt:407)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.notifyError(CameraDeviceImpl.java:2077)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks.lambda$Sm85frAzwGZVMAK-NE_gwckYXVQ(Unknown Source:0)
        at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$CameraDeviceCallbacks$Sm85frAzwGZVMAK-NE_gwckYXVQ.accept(Unknown Source:8)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:271)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:195)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:86)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:224)
        at android.os.HandlerThread.run(HandlerThread.java:67)

and here is the screenshot:
[image]

maxakash, 2020-09-16:
@milap612 I am also facing the black screen preview even for image capture on some devices.
But if the app is made to go to the background and then made to return to foreground, it starts working.
Can you try this. If this is also the issue with you, I think it is related to binding to lifecycle

milap612, 2020-09-17:
still the same result, tried background and return to foreground.
@maxakash on which device and what configuration (fps and resolution) you able to see with this scenario ?

maxakash, 2020-09-17:
> still the same result, tried background and return to foreground.
> @maxakash on which device and what configuration (fps and resolution) you able to see with this scenario ?

I tried with several Xiaomi devices, but mine issue was a bit different. My app was showing black screen while taking a picture on some devices.
As I don't have access to those devices , I cannot debug the issue. Also, crashlytics doesn't show any such issue.

utsavDave97, 2021-03-25:
> I tried with several Xiaomi devices, but mine issue was a bit different. My app was showing black screen while taking a picture on some devices.
> As I don't have access to those devices , I cannot [...]

Any fix for this @maxakash

maxakash, 2021-03-26:
I think updating the Library to the latest version fixed it.
I fixed it but I am not sure this was the fix.

NomadCR34, 2021-08-06:
> I think updating the Library to the latest version fixed it. I fixed it but I am not sure this was the fix.
would you say which library did you update?
I am facing same problem and i cant fixed it

NomadCR34, 2021-08-06:
would you say which library did you update?
I am facing same problem and i cant fixed it

zokica8, 2021-09-27:
Hello everyone. I had the issue of running the app on xiaomi mi 10 phone which has an Android 11 operating system. When I tried to run the app on xiaomi pocophone f1 with Android 10 operating system, it worked. When the app fails on mi 10 it tried to open the camera but it shows only the black screen and the onError callback of CameraDevice.StateCallback() is called everytime. 

What I tried to do next was to get a bug report for both phones and the results are the following:

on xiaomi mi 10 this is the most important line: 

I CameraManagerGlobal: Connecting to camera service
W ServiceManager: Permission failure: **android.permission.CAMERA_OPEN_CLOSE_LISTENER** from uid=10224 pid=13211

on pocophone f1 I got the list of cameras:

I CameraManagerGlobal: Connecting to camera service
D CAM_Camera2CompatAdapterCommon: All available camera ids: [0, 1, 2, 3, 4, 5]
D CAM_Camera2CompatAdapterCommon: ====================================================================
D CAM_Camera2CompatAdapterCommon:  BACK: [main, aux, sat, bokeh, virtual, infrared] = [0, 2, 3, 4, -1, -1]
D CAM_Camera2CompatAdapterCommon: FRONT: [main, aux, sat, bokeh, virtual, infrared] = [1, 5, -1, -1, -1, -1]
D CAM_Camera2CompatAdapterCommon: ====================================================================
D CAM_Camera2CompatAdapterCommon: X: init()

If needed I could provide the bug report text files, wanted to ask why is the app failing to open the camera on Android 11 phones?

One more note:  the onError callback on Android 11 phones is called right after this method call in `initializeCamera()` when setting up the preview for camera: 

`session.setRepeatingBurst(previewRequestList, null, cameraHandler)`

session object is of type CameraConstrainedHighSpeedCaptureSession.

Does anybody know why this could be an issue? 
Thanks in advance for the answer.

zoran-vasilic, 2022-04-05:
@maxakash which library did you fix?

zokica8, 2022-04-06:
Hi. I still have the issue with running this example of capturing a slow motion video on my phone, which is Mi 11T Pro. When I created the bug report for the whole device, I found two lines that could be the reason why this is not working.

ServiceManager: Permission failure: android.permission.SYSTEM_CAMERA from uid=somenumber pid=somenumber
ServiceManager: Permission failure: [android.permission.CAMERA](https://developer.android.google.cn/reference/android/Manifest.permission#CAMERA)_OPEN_CLOSE_LISTENER from uid=somenumber pid=somenumber

On Android 10 and lower devices it works, from Android 11 it does not work.
The only thing that I found was that from Android 11, the OS introduced the system camera permission, which is assigned at the system level. This is from the Android documentation:

"On devices running Android 11 or higher, the Android framework supports system cameras, which are camera devices that are visible only to processes with the android.permission.SYSTEM_CAMERA permission and regular camera permissions. The android.permission.SYSTEM_CAMERA permission, introduced in Android 11, has a protection level of system|signature. This means that only apps installed on the system partition with or signed with the same certificate as the system can be granted this permission. Third-party public apps can't access system cameras.

System cameras are useful for device manufacturers that want to implement features that require access to a camera but are also restricted to privileged or system apps.

Because systems apps that are granted the android.permission.SYSTEM_CAMERA permission must also have the android.permission.CAMERA permission, users can choose to revoke normal CAMERA permissions to prevent such an app from accessing the cameras on the device."

Basically what this means is that the only way you can create this kind of an app and run this example is for this app to be a system app, an app that is part of a system, like most Google apps for most Android phones that you can not uninstall from the device, unless you want to root the device.

Is there anybody who has experience in creating these kind of apps?
Also, I found another link that could be useful.

https://blog.csdn.net/KLYAZWS/article/details/119376225