android/camera-samples #293: Camera2Basic: Captured image has more field of view as compared to camera preview on Pixel 3 simulator.

pawarprasad37, 2020-08-30:
[image]
[image]

Simulator config: AVD Pixel 3 API 28 (with play services) x86.

I tried out the  Camera2Basic sample and after clicking an image, the captured image has more field of view as compared to what was being displayed in the camera preview. 
From user perspective, I'm expecting the captured image to look exactly the same as preview.
Is this a bug or some configuration is required for this behavior to be enabled?

pawarprasad37, 2020-08-30:
Update: I see the same issue with my OnePlus 7 Pro running Android 10.

alexcohn, 2020-10-01:
Camera2Basic uses the [max total still picture size](https://github.com/android/camera-samples/blob/71e83b90d9ce25b74970e98aaf1c26532606eb2c/Camera2Basic/app/src/main/java/com/example/android/camera2/basic/fragments/CameraFragment.kt#L196), not even trying to match the aspect ratio of the preview, therefore the result depends on the device. 

But even if you choose the matching aspect ratio, the captured image will most likely be slightly different from the way you see preview on your screen, like in https://github.com/android/camera-samples/issues/302. The cause is that the preview surface is extended to fill the view using center-crop transform. To avoid this, you must shrink the preview view and there will be unused margins on the screen.

owahltinez, 2020-11-08:
As @alexcohn said, we are using center-crop for the sizing logic. If you want the same aspect ratio, you will have to either:
1. Crop the resulting image yourself
2. Use letter-box transformation instead

closed 2020-11-08 by owahltinez