android/camera-samples #294: A bad case about camerax

Oo121oO, 2020-09-01:
I found a device that cannot support camerax, also it's android 5.1.1 api 22.

Bellow are the details about the device:
Build.MANUFACTURER:  Vivo 
Build.BRAND: Vivo
Build.MODEL: Vivo Y51A
Build.VERSION.SDK_INT: 22
Build.VERSION.RELEASE: 5.1.1

Here are the things: （we test on CameraXBasic Demo）
1、first time I launch app, it preview very well. Once I click switch button, it totally gives me a dark screen, and nothing I can do except relaunch it.

2、first time I launch app, I click capture button, nothing change, no photo, no response.

please let me know if there are any progresses. thx.

owahltinez, 2020-09-14:
Thanks for the bug report! It sounds like a device-specific issue. Do you mind starting a thread in the [developer forums](https://g.co/camerax/developers) to alert the team about it?

Oo121oO, 2020-09-14:
Well，thx for your advice，I will send my application right now, I am appreciate to hear from them ASAP.

owahltinez, 2020-09-14:
I see that you already started a thread there, thank you! You should get a response promptly. I'll close this issue now so we can follow up there.

closed 2020-09-14 by owahltinez