android/camera-samples #295: CameraAccessException w/ Android R

ashenriques, 2020-09-01:
When running the sample [CameraXBasic](https://github.com/android/camera-samples/tree/master/CameraXBasic) on a device running Android R (in my case the latest version of the Android Studio emulator (30.0.26-6755297) ), the middle button of the App to take the picture does nothing. On logcat one can see on the first attempt the following:

`2020-09-01 23:10:59.159 5253-5283/com.android.example.cameraxbasic E/CameraCaptureSession: Session 0: Exception while stopping repeating: 
    android.hardware.camera2.CameraAccessException: CAMERA_ERROR (3): The camera device has encountered a serious error
        at android.hardware.camera2.impl.CameraDeviceImpl.checkIfCameraClosedOrInError(CameraDeviceImpl.java:2231)
        at android.hardware.camera2.impl.CameraDeviceImpl.stopRepeating(CameraDeviceImpl.java:1241)
        at android.hardware.camera2.impl.CameraCaptureSessionImpl.close(CameraCaptureSessionImpl.java:578)
        at androidx.camera.camera2.internal.CaptureSession.closeCameraCaptureSession(CaptureSession.java:572)
        at androidx.camera.camera2.internal.CaptureSession.release(CaptureSession.java:520)
        at androidx.camera.camera2.internal.Camera2CameraImpl.releaseSession(Camera2CameraImpl.java:493)
        at androidx.camera.camera2.internal.Camera2CameraImpl.resetCaptureSession(Camera2CameraImpl.java:1045)
        at androidx.camera.camera2.internal.Camera2CameraImpl.closeCamera(Camera2CameraImpl.java:410)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.reopenCameraAfterError(Camera2CameraImpl.java:1460)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.handleErrorOnOpen(Camera2CameraImpl.java:1434)
        at androidx.camera.camera2.internal.Camera2CameraImpl$StateCallback.onError(Camera2CameraImpl.java:1412)
        at androidx.camera.camera2.internal.CameraDeviceStateCallbacks$ComboDeviceStateCallback.onError(CameraDeviceStateCallbacks.java:119)
        at android.hardware.camera2.impl.CameraDeviceImpl.notifyError(CameraDeviceImpl.java:1629)
        at android.hardware.camera2.impl.CameraDeviceImpl.lambda$oDs27OTfKFfK18rUW2nQxxkPdV0(Unknown Source:0)
        at android.hardware.camera2.impl.-$$Lambda$CameraDeviceImpl$oDs27OTfKFfK18rUW2nQxxkPdV0.accept(Unknown Source:8)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.doInvoke(PooledLambdaImpl.java:278)
        at com.android.internal.util.function.pooled.PooledLambdaImpl.invoke(PooledLambdaImpl.java:201)
        at com.android.internal.util.function.pooled.OmniFunction.run(OmniFunction.java:97)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$1.run(SequentialExecutor.java:110)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.workOnQueue(SequentialExecutor.java:230)
        at androidx.camera.core.impl.utils.executor.SequentialExecutor$QueueWorker.run(SequentialExecutor.java:172)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)`

further attempts don't print an exception, but one can read on logcat the following:
`Unable to issue take picture. Re-queuing image capture request`

Interesting enough this only happens with the main camera, if the user switches to the front camera, the app behaves as expected.

owahltinez, 2020-09-10:
I think this was caused by a fault in the emulator, which has been fixed in the latest release. Can you please updated to the latest SDK and try again?

ashenriques, 2020-09-15:
Thanks! I can confirm that with the latest emulator version the problem was fixed.

closed 2020-09-15 by ashenriques