android/camera-samples #301: Camera2Video crash when recording with Back camera in Android 11

buntupana, 2020-09-14:
Tried with Android 11 and Pixel 3a emulator. Every time I try to record a video with back camera the app will crash. Rest of Android versions work as expected.

How to reproduce:
- Open the app
- Select any back camera resolution
- Record a video

**Output:**
`2020-09-14 12:43:40.172 13317-13370/com.android.example.camera2.video E/MediaRecorder: stop failed: -1007
2020-09-14 12:43:40.235 13317-13370/com.android.example.camera2.video E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-1
    Process: com.android.example.camera2.video, PID: 13317
    java.lang.RuntimeException: stop failed.
        at android.media.MediaRecorder.stop(Native Method)
        at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)`

buntupana, 2020-09-14:
Just noticed that if I change in emulator back camera preferences from `VirtualScene` to `Emulated` it just works. Not sure if it's a code problem or an emulator bug.

hho114, 2020-10-20:
I got the same issue with both android 9 for real device and android 11 for the emulator.

@buntupana How you change in emulator back camera preferences from VirtualScene to Emulated?

buntupana, 2020-11-05:
@hho114  sorry for the delay

AVD Manager -> Edit (Desired emulator) -> Show Advanced Settings -> Camera secction

owahltinez, 2020-11-08:
Emulator had some issues, which should be fixed now. Please update Android Studio and the emulator to the latest version.

closed 2020-11-08 by owahltinez