android/camera-samples #302: Preview image height and captured image height are different

maxakash, 2020-09-15:
When i click an image, the resulting image is longer than the image shown in camera preview.
This results in very bad user experience.
I am using 
``` 
   val metrics = DisplayMetrics().also { viewFinder.display.getRealMetrics(it) }
            val screenAspectRatio = aspectRatio(metrics.widthPixels, metrics.heightPixels)

            val preview = Preview.Builder()
                    .setTargetAspectRatio(screenAspectRatio)
                    .build()

            imageCapture = ImageCapture.Builder()
                    .setTargetAspectRatio(screenAspectRatio)
                    .build()

            val cameraSelector = CameraSelector.Builder()
                    .requireLensFacing(CameraSelector.LENS_FACING_BACK)
                    .build()
```

```
private fun aspectRatio(width: Int, height: Int): Int {
        val previewRatio = kotlin.math.max(width, height).toDouble() / min(width, height)
        if (kotlin.math.abs(previewRatio - RATIO_4_3_VALUE) <= kotlin.math.abs(previewRatio - RATIO_16_9_VALUE)) {
            return AspectRatio.RATIO_4_3
        }
        return AspectRatio.RATIO_16_9
    }
```

[image]

[image]

alexcohn, 2020-10-01:
This only seems so; the captured image is displayed with FIT_CENTER, while the preview is equivalent to CENTER_INSIDE.

owahltinez, 2020-10-01:
Correct, the `PreviewView` class is performing a center-crop transform whereas it sounds like you want a letter-box transform. I don't think that's supported by the library (yet), so there's nothing that can be done on the app side if you are using `PreviewView`. The only alternative you have is creating your own `SurfaceProvider` which appropriately transforms the preview according to your requirements.

closed 2020-10-01 by owahltinez