android/camera-samples #303: Black camera preview on some devices

maxakash, 2020-09-15:
I am using camerax in my app to capture images. But it is not working on some devices.

A black screen is shown, but clicking on capture buttons captures the image.
Also, if the app is brought back to foreground from recents, preview starts to work.
So, I think the issue is with binding the camera to lifecycle.

Here is my code

```
  private fun startCamera() {

        cameraExecutor = Executors.newSingleThreadExecutor()
        val cameraProviderFuture = ProcessCameraProvider.getInstance(this@TakeImage)

        cameraProviderFuture.addListener({
            cameraProvider = cameraProviderFuture.get()

            val metrics = DisplayMetrics().also { viewFinder.display.getRealMetrics(it) }
            val screenAspectRatio = aspectRatio(metrics.widthPixels, metrics.heightPixels)

            val preview = Preview.Builder()
                    .setTargetAspectRatio(screenAspectRatio)
                    .build()

            imageCapture = ImageCapture.Builder()
                    .setTargetAspectRatio(screenAspectRatio)
                    .build()

            val cameraSelector = CameraSelector.Builder()
                    .requireLensFacing(CameraSelector.LENS_FACING_BACK)
                    .build()

            cameraProvider.unbindAll()

            try {
                cameraProvider.bindToLifecycle(this, cameraSelector, preview, imageCapture)
                preview.setSurfaceProvider(viewFinder.createSurfaceProvider())

            } catch (exc: Exception) {
                Log.e(TAG, "Use case binding failed", exc)
            }

        }, ContextCompat.getMainExecutor(this))
    }
```

```
private fun takePhoto() {
        val imageCapture = imageCapture ?: return

        val photoFile = File(
                getExternalFilesDir("scantmp"),
                SimpleDateFormat(FILENAME_FORMAT, Locale.US
                ).format(System.currentTimeMillis()) + ".png")

        val outputOptions = ImageCapture
                .OutputFileOptions
                .Builder(photoFile)
                .build()

        imageCapture.takePicture(
                outputOptions, ContextCompat.getMainExecutor(this), object : ImageCapture.OnImageSavedCallback {
            override fun onError(exc: ImageCaptureException) {
                Log.e(TAG, "Photo capture failed: ${exc.message}", exc)
            }

            override fun onImageSaved(output: ImageCapture.OutputFileResults) {
                savedImageUri = Uri.fromFile(photoFile)
                flash.visibility = View.GONE
                closeCamera.visibility = View.GONE
                takeAgain.visibility = View.VISIBLE
                saveImage.visibility = View.VISIBLE
                imgCapture.visibility = View.INVISIBLE
                imageCaptured.setImageURI(Uri.fromFile(photoFile))
                imageCaptured.visibility = View.VISIBLE
                viewFinder.visibility = View.GONE

            }
        })
    }
```

vismay11, 2020-09-26:
Hi, Is this issue fixed? Do you have any solution for this? I am facing the same issue. The camera works fine when the app is installed via Android Studio but gives a black screen on generating signed apk.

My device is Redmi 8A

owahltinez, 2020-10-01:
The issue tracker is being used for problems with the unmodified samples. If you have questions about usage, please use tag `android-camerax` in StackOverflow or start a thread in the [developer forums](https://g.co/camerax/developers)

closed 2020-10-01 by owahltinez

owahltinez, 2020-10-01:
Thanks for confirming. Can you please start a thread in the [developer forums](https://g.co/camerax/developers) and specify the device and Android version?