android/camera-samples #305: CameraXBasic - Buttons onClick events are not triggered the very first time the app is launched

lllAlonsolll, 2020-09-21:
Long story short: You need to click anywhere in the screen once prior to have any click listener event to work.

Steps to reproduce:
- Start the Camera Activity from the scratch.
- Make sure the first thing you click within that screen is any of the control buttons (switch/capture/gallery).

Actual Result:
- Nothing happens, it is just like if you haven't even clicked anything.

Expected Result:
- The events should be triggered.

Some additional details:
- I tried commenting out the PreviewView and launch the same activity. The issue is not happening when I do that, so that might point out that it is something related to PreviewView.
- It happens for absolutely any kind of button: (Button/MaterialButton/AppCompatButton and so on). I haven't tried with any other kind of views, but I suspect the result would end up the same.
- Once you have clicked once anywhere, all the events start working as expected.

owahltinez, 2020-10-02:
Thanks for reporting this! Are you seeing it in physical devices, or is this observed in an emulator?

lllAlonsolll, 2020-10-05:
@owahltinez I observed this in emulators, haven't tested out in physical devices, but I can give it a try and specify whether it could be reproduced that way or not.