android/camera-samples #308: oppo A37t 6.0 crash with Camera2Basic demo

zhenggy, 2020-09-24:
**Steps to reproduce： open app ->  home key to exit**

09-24 11:38:20.400 27612-27707/com.android.example.camera2.basic E/CameraFragment: Camera 0 error: (1) Camera in use
    java.lang.RuntimeException: Camera 0 error: (1) Camera in use
        at com.example.android.camera2.basic.fragments.CameraFragment$openCamera$$inlined$suspendCancellableCoroutine$lambda$1.onError(CameraFragment.kt:277)
        at android.hardware.camera2.impl.CameraDeviceImpl$CameraDeviceCallbacks$1.run(CameraDeviceImpl.java:1670)
        at android.os.Handler.handleCallback(Handler.java:815)
        at android.os.Handler.dispatchMessage(Handler.java:104)
        at android.os.Looper.loop(Looper.java:238)
        at android.os.HandlerThread.run(HandlerThread.java:61)

owahltinez, 2020-10-01:
I'm unable to reproduce this problem. Does it happen every time? What device are you seeing this on? Does it keep happening after your restart the device?

owahltinez, 2020-11-08:
Closing due to inactivity

closed 2020-11-08 by owahltinez