android/camera-samples #311: Misleading comment, revoke permissions stops app

sproctor, 2020-09-25:
I don't believe that permissions can be revoked while the app is paused. I did some testing, but I can't find documentation stating one way or the other. For me, apps are always stopped when permissions are changed, so permissions never need to be checked on resume, contrary to what is stated here: https://github.com/android/camera-samples/blob/master/CameraXBasic/app/src/main/java/com/android/example/cameraxbasic/fragments/CameraFragment.kt#L147

owahltinez, 2020-10-01:
Thanks for opening this issue!

I believe revoking app permissions while the app is paused might be possible in a multi-window or multi-screen environment. Also, since the app does not override `onStart()`, this would be the best place to do check for the permissions.

sproctor, 2020-10-02:
I've done a bit of testing. Revoking permissions in a multi-window environment closes the app. At least on android 8.1.

The only counter example I can find is this: https://commonsware.com/blog/2015/11/24/securityexceptions-runtime-permissions-reset-app-preferences.html For me, the app is still terminated when I hit reset app permissions. Maybe this behavior was changed sometime by 8.1.

owahltinez, 2020-10-02:
I don't think the behavior changed in 8.1, I think you found a bug in either the sample, or in the CameraX library :-)

My guess is that the camera device disconnection is not being handled appropriately, and that results in an error. What do you see in logcat when you revoke the permissions and the app crashes?

sproctor, 2020-10-02:
There is nothing in logcat when I reset app permission. The process is just terminated.

owahltinez, 2020-11-08:
That might be device-dependent behavior then. I don't think that there are any guarantees about the process restarting, and we recommend that developers should try to handle permissions during `onPause()`.

closed 2020-11-08 by owahltinez

sproctor, 2020-11-09:
That makes sense, except the docs say that you should check that you have the permission every time you do a dangerous operation. The operation happens in onViewCreated/onConfigurationChanged, not in onResume. My assumption is that the process is getting killed for having a running camera binding when the camera permission is revoked and that the check should be just before the camera operations.

https://developer.android.com/training/permissions/requesting#perm-check