android/camera-samples #314: CameraXBasic imageCapture.takePicture does not return any callback on my android 10 emulator

tomasmaks, 2020-09-30:
I used CameraXBasic in my production app for more than several months, everything was fine until 2 or 3 weeks ago takePicture function stopped returning any callback. I tried to update version and many different ways to implement CameraXBasic on my app, but it doesn't return any callback anymore. Both onError and onImageSaved are not returned. 

imageCapture.takePicture(outputOptions, cameraExecutor, object : ImageCapture.OnImageSavedCallback {
                    override fun onError(exc: ImageCaptureException) {
                        Log.e(TAG, "Photo capture failed: ${exc.message}", exc)
                    }

                    override fun onImageSaved(output: ImageCapture.OutputFileResults) {
                    })

Tried versions:
def camerax_version = '1.0.0-beta08'
def camerax_version = "1.0.0-beta10"
def camerax_version = "1.0.0-alpha04"

It seems it doesn't work on many  other Android 10 devices either, but works well on Android 9.

tomasmaks, 2020-10-02:
It still does not respond on Pixel 2 Android 10

owahltinez, 2020-10-02:
Have you tried using the unmodified sample?

tomasmaks, 2020-10-02:
What do you mean unmodified sample?

tomasmaks, 2020-10-07:
If by saying unmodified sample, you mean Camera2Basic, I tried to launch your sample app and it does not work on my emulator too.

owahltinez, 2020-11-08:
This was an emulator issue, it should be fixed now. Please update Android Studio and your emulator to the latest version and this issue should go away.

closed 2020-11-08 by owahltinez