android/camera-samples #320: Camera2Video record with overlay

patolax, 2020-10-11:
I am using the Camera2Video sample. I can record a basic camera preview output using this sample.

However, I want to draw something on top of the preview and then include that overlay in the final video captured in realtime (no post-processing). Is there a method to do this?

alexcohn, 2020-10-14:
Essentially, this is what the HDR example does.

patolax, 2020-10-15:
My phone camera does not have the minimum capabilities.  But I was able to get it to work.
It does not record anything to a file though.

Is there a sample where I can capture the camera output with say a watermark image overlaid on top of it?
So in realtime camera records with the watermark and then we can directly play the recorded video from the disk.

owahltinez, 2020-10-18:
@patolax this might be a better question for StackOverflow.

Here's an example of an activity which overlays some graphic on top of the camera feed, although it uses CameraX: https://github.com/owahltinez/androidx-camera-activity

closed 2020-10-18 by owahltinez

patolax, 2020-10-18:
I think you misunderstood. I want the overlay to be included in the capture. Your sample crops the capture. Regardless I want to capture video. Photos can be done easily using a canvas.

I have been searching for two weeks now, unfortunately, did not find anything yet. Yet, TikTok and Snapchat have done it years ago. Maybe it requires a big engineering team to do it. 
My scenario is simple, capture a video with a filter or with an extra dynamic graphic drawn on it.

longquangpham90, 2023-05-14:
I think cameraX not support add watermark realtime 
we need to add features