android/camera-samples #321: rotation issue of saved image

klower, 2020-10-12:
steps:
1. take photo in horizotal mode and save image
2. the image display horizontal

how to display vertical image like system camera?
[image]
I need to display like this below:
[image]

alexcohn, 2020-10-14:
Usually, the EXIF orientation flag will be written as part of the Jpeg header. The viewer can read the flag and apply the necessary rotation.

klower, 2020-10-15:
So how can I save EXIF into the captured image?

alexcohn, 2020-10-15:
You can use [ExifInterface](https://developer.android.com/reference/androidx/exifinterface/media/ExifInterface).

owahltinez, 2020-10-17:
As @alexcohn indicates, it seems that the issue is on the way the image is displayed. If you use Glide / Koil, images are rotated appropriately according to the EXIF metadata.

closed 2020-10-17 by owahltinez