android/camera-samples #329: Camera2Video stop failed on certain arguments

kwony, 2020-11-04:
Hello I am testing Camera2Video source code and found error as below

```
2020-11-04 14:55:39.317 30311-3794/com.android.example.camera2.video E/MediaRecorder: stop failed: -1007
2020-11-04 14:55:39.348 30311-3794/com.android.example.camera2.video E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-1
    Process: com.android.example.camera2.video, PID: 30311
    java.lang.RuntimeException: stop failed.
        at android.media.MediaRecorder._stop(Native Method)
        at android.media.MediaRecorder.stop(MediaRecorder.java:1472)
        at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
```

However when i change camera device session request configuration like below, error does not occur anymore. I am wondering there is any problem with `set(CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE, Range(args.fps, args.fps))`.

```
+++ b/Camera2Video/app/src/main/java/com/example/android/camera2/video/fragments/CameraFragment.kt
@@ -21,7 +21,11 @@ import android.content.Context
 import android.content.Intent
 import android.media.MediaScannerConnection
@@ -162,7 +166,7 @@ class CameraFragment : Fragment() {
             addTarget(viewFinder.holder.surface)
             addTarget(recorderSurface)
             // Sets user requested FPS for all targets
+           set(CaptureRequest.CONTROL_MODE, CameraMetadata.CONTROL_MODE_AUTO)
-            set(CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE, Range(args.fps, args.fps))
         }.build()
     }

```

This is my device specification 

- model number: SM-N971N

owahltinez, 2020-11-08:
Thanks for reporting this issue. Have you tried performing a long video recording? If you keep the record button pressed for 10-15 seconds, does the issue persist?

ilkerfindik, 2021-01-25:
@owahltinez Hi! I was checking out the issues for this issue and bunch of them are closed for inactivity and some for being emulators. I have this same issue and my details are as below.
- I have no change on the repo
- My device is Xiaomi Redmi Note 8 Pro 128 GB with plenty empty space
- Here are some results:
--- 🔴 4640 x 3472 20 FPS  
--- 🔴 4624 x 3472 20 FPS  -- 4640 vs. 4624 is not typo
--- 🔴 4624 x 2600 20 FPS
--- 🔴 4624 x 2135 20 FPS
--- 🟢 3840 x 2160 30 FPS 
--- 🔴 3472 x 3472 20 FPS 
--- 🔴 2560 x 1920 20 FPS
--- 🟢 1920 x 1440 20 FPS
--- 🟢 2340 x 1080 30 FPS
--- 🟢 1920 x 1080 60 FPS
--- 🟢 1440 x 1080 60 FPS
-  rest is all green with front camera all green

- I have only tried a couple of red ones about long videos, but failures continue to happen when I hold down for 15 seconds as well.

ilkerfindik, 2021-01-25:
Here are the listed options on my device because i had enough of typing them all

[image]
[image]

nedols, 2021-04-06:
Same issue on my  S21.

Here some results:
--- 🟢 Camera(0) 4032x3024 30 FPS, cameraId=0, size=4032x3024, fps=30
--- 🟢 Camera(0) 4032x2268 30 FPS, cameraId=0, size=4032x2268, fps=30
--- 🟢 Camera(0) 4032x1816 30 FPS, cameraId=0, size=4032x1816, fps=30
--- 🟢 Camera(0) 3024x3024 30 FPS, cameraId=0, size=3024x3024, fps=30
--- 🟢 Camera(0) 3840x2160 30 FPS, cameraId=0, size=3840x2160, fps=30
--- 🔴 Camera(0) 2400x1080 60 FPS, cameraId=0, size=2400x1080, fps=60
--- 🔴 Camera(0) 1920x1080 60 FPS, cameraId=0, size=1920x1080, fps=60
--- 🔴 Camera(0) 1920x864 60 FPS, cameraId=0, size=1920x864, fps=60
--- 🔴 Camera(0) 1440x1080 60 FPS, cameraId=0, size=1440x1080, fps=60
--- 🔴 Camera(0) 1088x1088 60 FPS, cameraId=0, size=1088x1088, fps=60
--- 🔴 Camera(0) 1280x720 60 FPS, cameraId=0, size=1280x720, fps=60
--- 🔴 Camera(0) 960x720 60 FPS, cameraId=0, size=960x720, fps=60
--- 🔴 Camera(0) 720x480 60 FPS, cameraId=0, size=720x480, fps=60
--- 🔴 Camera(0) 640x480 60 FPS, cameraId=0, size=640x480, fps=60
--- 🔴 Camera(0) 640x360 60 FPS, cameraId=0, size=640x360, fps=60
--- 🔴 Camera(0) 352x288 60 FPS, cameraId=0, size=352x288, fps=60
--- 🔴 Camera(0) 320x240 60 FPS, cameraId=0, size=320x240, fps=60
--- 🔴 Camera(0) 256x144 60 FPS, cameraId=0, size=256x144, fps=60
--- 🔴 Camera(0) 176x144 60 FPS, cameraId=0, size=176x144, fps=60
--- 🟢 Camera(2) 4032x3024 30 FPS, cameraId=2, size=4032x3024, fps=30
--- 🟢 Camera(2) 4032x2268 30 FPS, cameraId=2, size=4032x2268, fps=30
--- 🟢 Camera(2) 4032x1816 30 FPS, cameraId=2, size=4032x1816, fps=30
--- 🟢 Camera(2) 3024x3024 30 FPS, cameraId=2, size=3024x3024, fps=30
--- 🟢 Camera(2) 3840x2160 30 FPS, cameraId=2, size=3840x2160, fps=30
--- 🔴 Camera(2) 2400x1080 60 FPS, cameraId=2, size=2400x1080, fps=60
--- 🔴 Camera(2) 1920x1080 60 FPS, cameraId=2, size=1920x1080, fps=60
--- 🔴 Camera(2) 1920x864 60 FPS, cameraId=2, size=1920x864, fps=60
--- 🔴 Camera(2) 1440x1080 60 FPS, cameraId=2, size=1440x1080, fps=60
--- 🔴 Camera(2) 1088x1088 60 FPS, cameraId=2, size=1088x1088, fps=60
--- 🔴 Camera(2) 1280x720 60 FPS, cameraId=2, size=1280x720, fps=60
--- 🔴 Camera(2) 960x720 60 FPS, cameraId=2, size=960x720, fps=60
--- 🔴 Camera(2) 720x480 60 FPS, cameraId=2, size=720x480, fps=60
--- 🔴 Camera(2) 640x480 60 FPS, cameraId=2, size=640x480, fps=60
--- 🔴 Camera(2) 640x360 60 FPS, cameraId=2, size=640x360, fps=60
--- 🔴 Camera(2) 352x288 60 FPS, cameraId=2, size=352x288, fps=60
--- 🔴 Camera(2) 320x240 60 FPS, cameraId=2, size=320x240, fps=60
--- 🔴 Camera(2) 256x144 60 FPS, cameraId=2, size=256x144, fps=60
--- 🔴 Camera(2) 176x144 60 FPS, cameraId=2, size=176x144, fps=60

If i remove `CaptureRequest.CONTROL_AE_TARGET_FPS_RANGE` configuration from **CaptureFragment** i can record video, with all listed configurations.

I notice that the video recorded with one of the 🔴 configurations, are like a slowmotion videos. This happens only for those configurations.