android/camera-samples #335: CameraXBasic: on legacy Samsung: Saves images to root directory and gallery never updates

dkati, 2020-11-21:
Good evening. As title says:

CameraXBasic example saves the image to
/data/data/com.dialog.myapplication/files/2020-11-21-12-39-56-595.jpg
and photo never shows up at gallery.

Samsung S4, android lollipop (5.0.1 sdk 21)

owahltinez, 2020-11-24:
Hi @dkati, the images show in the default gallery application on the devices I have tested. The images themselves are not rotated, they use EXIF metadata and then the photo viewer must rotate the image. Again, the gallery application is responsible for that.

I would suggest trying a different gallery application, please reopen if the issue persists.

closed 2020-11-24 by owahltinez

dkati, 2020-11-24:
The "saves to root directory" is unrelated to gallery though

owahltinez, 2020-11-25:
The choice of output directory is deliberate: https://github.com/android/camera-samples/blob/main/CameraXBasic/app/src/main/java/com/android/example/cameraxbasic/MainActivity.kt#L70