android/camera-samples #342: Screen Flashing when Recording in Camera2Video

mdsheraj123, 2020-12-26:
When testing on an emulator and any device really, the screen flashes when we record video. I understand it is due to the repeating animation but it is not natural behaviour and looks like a bug along with taking up system resources to show the animation during an intensive task such as recording video.
I suggest there to be a flash at the start and end of recording video and no repeating animation.

Device model: sdk_gphone_x86_arm
Build number: sdk_gphone_x86_arm-userdebug 11 RSR1.201013.001 6903271 dev-keys

It's not an error so not attaching any logs.

emersonk, 2020-12-29:
I am having this 'issue' as well. As you said, it's not an error, but it certainly doesn't look great for official sample code.
I am testing on an LG-H933 running Android 9.

amazingvoice, 2021-04-19:
Got the same problem here and didn't event notice it was because of an animation... Thanks for the info