android/camera-samples #344: Aspect ratio not considering custom sized views

mrousavy, 2021-01-04:
Hi!

I'm developing an abstraction over the Camera2 API and found this sample to be very useful, so thanks!

I'm currently stuck with one issue though, and I can't seem to resolve that (since 2 days now), I keep finding old Camera related S/O threads and no answers in this repo's Issues/PRs.

My Preview (SurfaceView/TextureView) is stretched. Here's how it looks:

<table>
<tr>
<th>
Preview (using the `AutoFitSurfaceView`)
</th>
<th>
Actual Photo with resizeMode/scaleType "contain"
</th>
</tr>
<tr>
<td width="50%">
[image]
</td>
<td width="50%">
[image]
</td>
</tr>
</table>

as you can see, the first image is really weirdly stretched while the photo being shot is actually in correct dimensions.
I'm _pretty_ sure that the cause of the problem is the `AutoFitSurfaceView` which only uses the display size for aspect ratio calculation, but that obviously doesn't work if the `SurfaceView` is not the same size as the phone display, as seen in my screenshot. (Bottom Bar & Status Bar takes away some space)

I thought maybe someone here would be so kind to help me out here. 😄 

Thanks!

mrousavy, 2021-01-12:
cc maybe @owahltinez @zhaonian and @panabuntu 🙏

buntupana, 2021-01-12:
Hi, the xml layout would help

mrousavy, 2021-01-13:
Hi @buntupana!

Since I'm working with a different renderer (React Native) I don't have an XML Layout. This should not make a difference tho, as I just create everything programmatically.

Here's my `CameraView` which I then simply render at whatever size:

```kt
class CameraView(context: Context) : RelativeLayout(context), LifecycleEventListener {
    internal var cameraThread: HandlerThread? = null
    internal var cameraHandler: Handler? = null
    internal var cameraDevice: CameraDevice? = null
    internal val cameraExecutor: Executor by lazy {
        Executors.newSingleThreadExecutor()
    }

    internal var imageReader: ImageReader? = null
    internal var imageThread: HandlerThread? = null
    internal var imageHandler: Handler? = null

    internal var captureSession: CameraCaptureSession? = null
    internal var characteristics: CameraCharacteristics? = null

    internal val surfaceView: AutoFitSurfaceView

    internal val cameraManager: CameraManager by lazy {
        context.applicationContext.getSystemService(Context.CAMERA_SERVICE) as CameraManager
    }
    private var previewCaptureRequest: CaptureRequest.Builder? = null

    private val lifecycleExecutor = Executors.newSingleThreadExecutor().asCoroutineDispatcher()
    private val cameraMutex = Mutex()
    private var isHostActive = false
    private val isCameraLifecycleActive: Boolean
        get() = isActive && isHostActive && isAttachedToWindow

    init {
        surfaceView = AutoFitSurfaceView(context).apply {
            layoutParams = LayoutParams(LayoutParams.MATCH_PARENT, LayoutParams.MATCH_PARENT)
        }
        viewFinder.holder.addCallback(object : SurfaceHolder.Callback {
            override fun surfaceDestroyed(holder: SurfaceHolder) = Unit

            override fun surfaceChanged(
                    holder: SurfaceHolder,
                    format: Int,
                    width: Int,
                    height: Int) = Unit

            override fun surfaceCreated(holder: SurfaceHolder) {
                val previewSize = getPreviewOutputSize(surfaceView.display, characteristics!!, SurfaceHolder::class.java)
                surfaceView.setAspectRatio(previewSize.width, previewSize.height)

                surfaceView.post { 
                    configureSession()
                    configurePreview()
                }
            }
        })
        addView(surfaceView)
    }

    override fun onAttachedToWindow() {
        super.onAttachedToWindow()
        Log.i(REACT_CLASS, "onAttachedToWindow() called")
        GlobalScope.launch {
            configureSession()
        }
    }

    /**
[... 142 lines cut ...]
                previewCaptureRequest!!.set(CaptureRequest.CONTROL_AE_MODE, CaptureRequest.CONTROL_AE_MODE_ON)
            } else {
                previewCaptureRequest!!.set(CaptureRequest.FLASH_MODE, CaptureRequest.FLASH_MODE_OFF)
                previewCaptureRequest!!.set(CaptureRequest.CONTROL_AE_MODE, null)
            }

            if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M && captureSession is CameraConstrainedHighSpeedCaptureSession) {
                // The camera session is set up as a High-Speed capture session, so we want to request a high-speed preview.
                val session = captureSession as CameraConstrainedHighSpeedCaptureSession
                val highSpeedCaptureRequests = session.createHighSpeedRequestList(previewCaptureRequest!!.build())
                session.setRepeatingBurst(highSpeedCaptureRequests, null, cameraHandler!!)
            } else {
                // The camera session is set up as a normal capture session, preview with normal FPS range to the surface
                captureSession!!.setRepeatingRequest(previewCaptureRequest!!.build(), null, cameraHandler!!)
            }
        } else {
            captureSession?.stopRepeating()
        }
    }

    // ...
}
```

(Note: For the sake of demonstrating the concept, I have removed quite a bit of logic from this revolving around locking and synchronization, in this example the SurfaceView gets created and already assumes that a session has been configured (characteristics are not null))

buntupana, 2021-01-13:
@mrousavy did you try to replicate it in an Android Native App?

mrousavy, 2021-01-13:
@buntupana no, since that sizes the `AutoFitSurfaceView` to full-screen. I'm assuming if you add a bottom margin, it would look stretched too, I'll test that as soon as I can.

buntupana, 2021-01-13:
The problem with your view is that the SurfaceView is not cropping center. Check in your AutoFitSurfaceView what are the values of `width` `height` and what are the values of the result `newWidth` and `newHeight`...these last ones should be equal or bigger than the original ones

lutean, 2021-09-24:
I had the same problem. In my case solution is to change the root layout to FrameLayout. And AutoFitSurfaceView should be a child of the root.

robin2046, 2021-12-27:
> I had the same problem. In my case solution was to change the root layout to `FrameLayout`. And `AutoFitSurfaceView` should be a child of the root. Previously root was `ConstraintLayout`.

This bothered me for the whole afternoon trying to figure out why ConstraintLayout prevents the cropping of the surface view, any idea of the reason?