android/camera-samples #348: Camera2Video glitching and green flickering in recorded video in Galaxy S6

buntupana, 2021-01-09:
In Samsung Galaxy S6 the video result has a green flickering and glitching effect

[Gif of the issue](https://media.giphy.com/media/Ssj9BpyBZTea49NIa9/giphy.gif)

buntupana, 2021-02-03:
@owahltinez any idea why this is happening?

HasimD, 2021-02-18:
Yes. I am experiencing this problem too.

buntupana, 2021-02-18:
@HasimD the only solution I found is change my implementation to use TextureView instead SurfaceView...just if it's useful for you. It's a bit of shame that Google doesn't provide a good examples and don't even attend the tickets created

HasimD, 2021-02-18:
I agree with you,  i'm very surprised. The offical sample doesn't seem like professional. Also if I click to the capture button, the app crashes. I need to press it for a while. They said it is not an error and they will change it. Weird.

nedols, 2021-04-06:
The app c

> 
> 
> I agree with you, i'm very surprised. The offical sample doesn't seem like professional. Also if I click to the capture button, the app crashes. I need to press it for a while. They said it is not an error and they will change it. Weird.

Looking at the code this is the correct behaviour. 
To start recording you need to tap&hold the capture button. When you release the capture button the application stop to recording video.

amazingvoice, 2021-04-19:
I'm experiencing the same problem on my Galaxy s7 and Pixel 3. This is the only offical video recording demo using Camera2, which is embarrassing...

amazingvoice, 2021-04-19:
@buntupana I just happened to find that an animation causes the white flash. Search for the variable _animationTask_ in _CameraFragment.kt_. Comment that out will solve the problem.

buntupana, 2021-04-19:
@amazingvoice I'm not complaining about the white flickering, it's the green flickering and the glitching when the video is recorded what this ticket is about