android/camera-samples #350: CameraX Back and front camera con't preview at same time ?

Reginer, 2021-01-15:
```
var cameraBack = cameraProvider.bindToLifecycle(this, cameraSelectorBack, previewBack)
var cameraFront = cameraProvider.bindToLifecycle(this, cameraSelectorFront, previewFront)
```
```
 java.lang.IllegalArgumentException: Multiple LifecycleCameras with use cases are registered to the same LifecycleOwner.
        at androidx.camera.lifecycle.LifecycleCameraRepository.bindToLifecycleCamera(LifecycleCameraRepository.java:269)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:423)
        at androidx.camera.lifecycle.ProcessCameraProvider.bindToLifecycle(ProcessCameraProvider.java:274)
```

agent10, 2021-04-14:
It seems it's not supported. https://issuetracker.google.com/issues/151866349

Reginer, 2021-04-15:
I get it

letuss004, 2021-12-28:
@agent10 Do u have any solution cuz I have exactly same problem.

agent10, 2021-12-28:
@letuss004 I do) I use pure Camera2 api

letuss004, 2021-12-29:
@agent10 haha i expected an camerax but camera2 fair enough. Did u share source in github? I really want to take a look

gokor8, 2023-04-04:
@agent10 what do you do with devices that are not supported by hardware both cameras? As far as I understand, not all devices support both camera in time

Reginer, 2023-04-04:
> @agent10 what do you do with devices that are not supported by hardware both cameras? As far as I understand, not all devices support both camera in time
As far as I understand, all devices support both camera same time use Camera1 api.

kailianc, 2023-04-04:
Hi, CameraX will support back and front camera operating at the same time in the next alpha release. Not only for preview but also for image capture and potentially other use cases as well. We will appreciate if you could try it and give feedback to us.

kailianc, 2023-04-04:
To check the capability of concurrent cameras in Camera2, you can use FEATURE_CAMERA_CONCURRENT.

https://developer.android.com/reference/android/content/pm/PackageManager#FEATURE_CAMERA_CONCURRENT

lodgkk, 2023-05-11:
> Hi, CameraX will support back and front camera operating at the same time in the next alpha release. Not only for preview but also for image capture and potentially other use cases as well. We will appreciate if you could try it and give feedback to us.

Has the alpha version that supports using both front and back cameras simultaneously been released? What is the version number?

Reginer, 2023-05-11:
> Hi, CameraX will support back and front camera operating at the same time in the next alpha release. Not only for preview but also for image capture and potentially other use cases as well. We will appreciate if you could try it and give feedback to us.

Yes, of course .

kailianc, 2023-05-11:
Hi lodgkk,

Version 1.3.0-alpha06 is the latest version supporting both front and back cameras simultaneously.
https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha06

This is the javadoc for main api:
https://android.googlesource.com/platform/frameworks/support/+/refs/heads/androidx-main/camera/camera-lifecycle/src/main/java/androidx/camera/lifecycle/ProcessCameraProvider.java#759

This is the sample app to show a front + back camera previews in Picture-in-Picutre layout or Side-by-Side layout: https://android.googlesource.com/platform/frameworks/support/+/refs/heads/androidx-main/camera/integration-tests/coretestapp/src/main/java/androidx/camera/integration/core/ConcurrentCameraActivity.java#239

Let us know if any other questions.

lodgkk, 2023-11-17:
> Hi lodgkk,
>
> Version 1.3.0-alpha06 is the latest version supporting both front and back cameras simultaneously. https://developer.android.com/jetpack/androidx/releases/camera#1.3.0-alpha06
>
> This is [...]

Concurrent camera is not supported on the device ,How can I solve this?

oan001, 2024-03-19:
I am getting this error when i call cameraProvider.getAvailableConcurrentCameraInfos(). I tested this on two devices it works fine on samsung note 10+ but gives this error on sony xperia 1 mark 3. 

this is the code:
```
for (List<CameraInfo> cameraInfoList : cameraProvider.getAvailableConcurrentCameraInfos()) {
            for (CameraInfo cameraInfo : cameraInfoList) {
                if (cameraInfo.getLensFacing() == CameraSelector.LENS_FACING_FRONT) {
                    cameraSelectorPrimary = cameraInfo.getCameraSelector();
                } else if (cameraInfo.getLensFacing() == CameraSelector.LENS_FACING_BACK) {
                    cameraSelectorSecondary = cameraInfo.getCameraSelector();
                }
            }
            if (cameraSelectorPrimary == null || cameraSelectorSecondary == null) {
                // If either a primary or secondary selector wasn't found, reset both
                // to move on to the next list of CameraInfos.
                cameraSelectorPrimary = null;
                cameraSelectorSecondary = null;
            } else {
                // If both primary and secondary camera selectors were found, we can
                // conclude the search.
                break;
            }
        }
```

FATAL EXCEPTION: main
                 Process: com.rbsol.project.lucided, PID: 10937
                 java.lang.IllegalArgumentException: No camera can be find for id: 5
                 	at androidx.camera.camera2.internal.concurrent.Camera2CameraCoordinator.lambda$createCameraSelectorById$0(Camera2CameraCoordinator.java:189)
                 	at androidx.camera.camera2.internal.concurrent.Camera2CameraCoordinator$$ExternalSyntheticLambda0.filter(Unknown Source:2)
                 	at androidx.camera.core.CameraSelector.filter(CameraSelector.java:128)
                 	at androidx.camera.lifecycle.ProcessCameraProvider.getCameraInfoFromCameraSelector(ProcessCameraProvider.java:882)
                 	at androidx.camera.lifecycle.ProcessCameraProvider.getAvailableConcurrentCameraInfos(ProcessCameraProvider.java:823)
                 	at com.rbsol.project.Fragments.CameraXFragment.bindToLifecycleForConcurrentCamera(CameraXFragment.java:551)
                 	at com.rbsol.project.Fragments.CameraXFragment.lambda$bindPreviewForPiP$7$com-rbsol-project-Fragments-CameraXFragment(CameraXFragment.java:510)
                 	at com.rbsol.project.Fragments.CameraXFragment$$ExternalSyntheticLambda5.run(Unknown Source:4)
                 	at android.os.Handler.handleCallback(Handler.java:942)
                 	at android.os.Handler.dispatchMessage(Handler.java:99)
                 	at android.os.Looper.loopOnce(Looper.java:346)
                 	at android.os.Looper.loop(Looper.java:475)
                 	at android.app.ActivityThread.main(ActivityThread.java:7950)
                 	at java.lang.reflect.Method.invoke(Native Method)
                 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
                 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:942)