android/camera-samples #351: When I opened the app for the second time it must have crashed

TwilightKHQ, 2021-01-19:
Can anyone help me to see why I have this problem, I have not modified any content.

device:
Pixel 3 Android 10 and Xiaomi Redmi K30 Ultra Android 11

logcat:
2021-01-19 09:13:05.870 353-353/com.android.example.cameraxbasic E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.android.example.cameraxbasic, PID: 353
    java.lang.RuntimeException: Unable to start activity ComponentInfo{com.android.example.cameraxbasic/com.android.example.cameraxbasic.MainActivity}: android.view.InflateException: Binary XML file line #17 in com.android.example.cameraxbasic:layout/activity_main: Binary XML file line #17 in com.android.example.cameraxbasic:layout/activity_main: Error inflating class androidx.fragment.app.FragmentContainerView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3270)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3409)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:83)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2016)
        at android.os.Handler.dispatchMessage(Handler.java:107)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7356)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:930)
     Caused by: android.view.InflateException: Binary XML file line #17 in com.android.example.cameraxbasic:layout/activity_main: Binary XML file line #17 in com.android.example.cameraxbasic:layout/activity_main: Error inflating class androidx.fragment.app.FragmentContainerView
     Caused by: android.view.InflateException: Binary XML file line #17 in com.android.example.cameraxbasic:layout/activity_main: Error inflating class androidx.fragment.app.FragmentContainerView
     Caused by: java.lang.IllegalArgumentException: ID does not reference a View inside this Activity

lijheng, 2021-01-30:
I had the same problem. And change 'Navigation.findNavController(requireActivity(), R.id.fragment_container).navigate(
                    PermissionsFragmentDirections.actionPermissionsToCamera())' in PermissionFragment to ' parentFragment?.findNavController()?.navigate(PermissionsFragmentDirections.actionPermissionsToCamera())‘ resolved it. 
 https://developer.android.com/guide/navigation/navigation-getting-started?hl=zh-cn#navigate

lijheng, 2021-01-31:
I get this crash. And I move PermissionsFragment.onCreate code to onActivityCreated() resolevd it.

TwilightKHQ, 2021-02-05:
[image]
@lijheng 谢了，老铁.

closed 2021-02-05 by TwilightKHQ