android/camera-samples #360: CameraX crashed at Samsung S10 5G  in "androidx.camera:camera-camera2:1.1.0-alpha02" version

DJSeokHo, 2021-03-05:
in cameraX API, we have this "cameraProvider = cameraProviderFuture.get()". this line will lead app to crash when use "androidx.camera:camera-camera2:1.1.0-alpha02" at Samsung S10 5G.

but "cameraProvider = cameraProviderFuture.get()" is good when I use "androidx.camera:camera-camera2:1.1.0-alpha01" at Samsung S10 5G.

so, the bug is:

In Samsung S10 5G:
cameraProvider = cameraProviderFuture.get() crash when "androidx.camera:camera-camera2:1.1.0-alpha02"
cameraProvider = cameraProviderFuture.get() is ok when "androidx.camera:camera-camera2:1.1.0-alpha01"

please confirm this bug and fix it. thanks.

joey-harward, 2021-03-06:
Was this ever resolved? I am seeing the same issue with Samsung Galaxy S20+ 5G.

mobee-bill, 2021-03-16:
I am seeing this issue as well with users on device LG V60:
Caused by java.util.concurrent.ExecutionException
androidx.camera.core.InitializationException: java.lang.RuntimeException: Error retrieving camcorder profile params

andreyvatoropinit, 2021-03-23:
i solved it on the next way. Choose max resolution before shooting and at moment of shooting press button and keep finger about 7 second.