android/camera-samples #362: Can it access usb camera? Does it has uvc support

bjain409, 2021-03-08:
Please follow the following instructions before filing a bug:

1. Github issues under this project are only for **sample**-related issues. If you have modified the sample code in any way, then this is probably not the appropriate place to report a bug.
1. If you are trying to report a bug related to the **CameraX library** then you can do so at the Android Issue Tracker: https://issuetracker.google.com/components/618491.
1. Include the specific sample where you found the issue, in addition to device model and specific Android build number -- which can be found under Settings > About phone.
1. Include error logs from logcat, if there are any.
1. If you can, also attach a [bug report](https://developer.android.com/studio/debug/bug-report)-- but please keep in mind that personal information can sometimes be found in those, so double check before attaching.

wwdok, 2021-03-23:
Why close this issue, i also wonder if cameraX has support USB camera, if not, when will it be OK

mrliuzhao, 2021-04-07:
I am also looking for Support for USB Camera in CameraX

luyanan123, 2021-04-12:
> 我也在寻找CameraX中对USB摄像头的支持

您好，麻烦请问一下您找到关于camerax连接usb相关的东西了吗？

mrliuzhao, 2021-04-14:
> > 我也在寻找CameraX中对USB摄像头的支持
> 
> 您好，麻烦请问一下您找到关于camerax连接usb相关的东西了吗？

我这边最近调查了一下，发现安卓Camera2中摄像头有3种：back, front, external。back和front就是前置、后置摄像头，external则是外部摄像头。有的安卓系统把连接的USB摄像头识别为前置、后置，摄像头ID都是数字，而有的安卓系统把USB摄像头识别为External，摄像头ID就是linux中的设备名称，比如/dev/video0。这貌似就跟具体安卓系统编译的时候的一些底层驱动设置有关，具体我也还没有深入了解。

CameraX目前只能识别前置和后置摄像头，因此安卓系统中如果将USB摄像头分类在前置或后置，就可以用CameraX打开

wwdok, 2021-04-15:
@mrliuzhao Thanks for sharing information. I have posted another issue [here](https://github.com/nikishovde/CameraX-EXTERNAL/issues/1) that discuss this topic too, you can have a look. So the question turns to how to set USB camera as front or back when compiling AOSP ?
BTW, i wonder if android official team whether have plan to support USB camera in the future ? If yes, when is it, Thank you ! @owahltinez

Qamar4P, 2021-04-30:
@mrliuzhao @wow @bjain409 It's up to vendors because they customize the camera HAL. 
Any know the list of Android phones which support USB Cameras?

Reginer, 2021-05-13:
mark

AnyLifeZLB, 2023-05-23:
4年了，CameraX还不支持 USB 摄像头

byqy, 2024-09-10:
> 4年了，CameraX还不支持 USB 摄像头

5年了，CameraX还不支持 USB 摄像头