android/camera-samples #363: Are you sure Camera2SlowMotion can run?

RenXGang, 2021-03-09:
Please follow the following instructions before filing a bug:
in Pix 3 and sunsang it doesn't work, will show many fps options,but when i selected anyone ,app will carsh

zokica8, 2021-09-27:
Hello everyone. I had the issue of running the app on xiaomi mi 10 phone which has an Android 11 operating system. When I tried to run the app on xiaomi pocophone f1 with Android 10 operating system, it worked. When the app fails on mi 10 it tried to open the camera but it shows only the black screen and the onError callback of CameraDevice.StateCallback() is called everytime.

What I tried to do next was to get a bug report for both phones and the results are the following:

on xiaomi mi 10 this is the most important line:

I CameraManagerGlobal: Connecting to camera service
W ServiceManager: Permission failure: android.permission.CAMERA_OPEN_CLOSE_LISTENER from uid=10224 pid=13211

on pocophone f1 I got the list of cameras:

I CameraManagerGlobal: Connecting to camera service
D CAM_Camera2CompatAdapterCommon: All available camera ids: [0, 1, 2, 3, 4, 5]
D CAM_Camera2CompatAdapterCommon: ====================================================================
D CAM_Camera2CompatAdapterCommon: BACK: [main, aux, sat, bokeh, virtual, infrared] = [0, 2, 3, 4, -1, -1]
D CAM_Camera2CompatAdapterCommon: FRONT: [main, aux, sat, bokeh, virtual, infrared] = [1, 5, -1, -1, -1, -1]
D CAM_Camera2CompatAdapterCommon: ====================================================================
D CAM_Camera2CompatAdapterCommon: X: init()

If needed I could provide the bug report text files, wanted to ask why is the app failing to open the camera on Android 11 phones?

One more note: the onError callback on Android 11 phones is called right after this method call in initializeCamera() when setting up the preview for camera:

session.setRepeatingBurst(previewRequestList, null, cameraHandler)

session object is of type CameraConstrainedHighSpeedCaptureSession.

Does anybody know why this could be an issue?
Thanks in advance for the answer.

zokica8, 2022-04-06:
Hi. I still have the issue with running this example of capturing a slow motion video on my phone, which is Mi 11T Pro. When I created the bug report for the whole device, I found two lines that could be the reason why this is not working.

ServiceManager: Permission failure: android.permission.SYSTEM_CAMERA from uid=somenumber pid=somenumber
ServiceManager: Permission failure: [android.permission.CAMERA](https://developer.android.google.cn/reference/android/Manifest.permission#CAMERA)_OPEN_CLOSE_LISTENER from uid=somenumber pid=somenumber

On Android 10 and lower devices it works, from Android 11 it does not work.
The only thing that I found was that from Android 11, the OS introduced the system camera permission, which is assigned at the system level. This is from the Android documentation:

"On devices running Android 11 or higher, the Android framework supports system cameras, which are camera devices that are visible only to processes with the android.permission.SYSTEM_CAMERA permission and regular camera permissions. The android.permission.SYSTEM_CAMERA permission, introduced in Android 11, has a protection level of system|signature. This means that only apps installed on the system partition with or signed with the same certificate as the system can be granted this permission. Third-party public apps can't access system cameras.

System cameras are useful for device manufacturers that want to implement features that require access to a camera but are also restricted to privileged or system apps.

Because systems apps that are granted the android.permission.SYSTEM_CAMERA permission must also have the android.permission.CAMERA permission, users can choose to revoke normal CAMERA permissions to prevent such an app from accessing the cameras on the device."

Basically what this means is that the only way you can create this kind of an app and run this example is for this app to be a system app, an app that is part of a system, like most Google apps for most Android phones that you can not uninstall from the device, unless you want to root the device.

Is there anybody who has experience in creating these kind of apps?
Also, I found another link that could be useful.

https://blog.csdn.net/KLYAZWS/article/details/119376225