android/camera-samples #377: java.lang.IndexOutOfBoundsException, because of YuvToRgbConverter.yuvToRgb

viktor-idenfy, 2021-04-18:
Hello, we use cameraX for the production and started to notice increasing crashes, because of the ** yuvToRgb**  (specifically _**imageToByteArray(image: Image, outputBuffer: ByteArray)**_) [YuvToRgbConverter](https://github.com/android/camera-samples/blob/main/CameraUtils/lib/src/main/java/com/example/android/camera/utils/YuvToRgbConverter.kt) starting camera x core beta version ~20. 

We use this class for converting Image from cameraX to Bitmap and it is an official recommendation from Google themselves. 

Perpahs you should update the code because it is not stable anymore? 

Here is the list of devices with the crash:
CUBOT X19
Samsung Galaxy S9+
HUAWEI Y7 2019
BQru-5514G

The full crash log:

`Fatal Exception: java.lang.IndexOutOfBoundsException
off=1382400, len=1920 out of bounds (size=1382407)
java.nio.Buffer.checkBounds (Buffer.java:559)
java.nio.DirectByteBuffer.get (DirectByteBuffer.java:234)
com.idenfy.idenfySdk.documentscamerasession.ui.utils.CustomBitmapConverter.imageToByteArray (CustomBitmapConverter.java:117)
com.idenfy.idenfySdk.documentscamerasession.ui.utils.CustomBitmapConverter.convertYuvToBitmap (CustomBitmapConverter.java:41)
com.idenfy.idenfySdk.documentscamerasession.ui.utils.CustomDocumentImageProcessor.analyze (CustomDocumentImageProcessor.java:76)
androidx.camera.core.ImageAnalysis.lambda$setAnalyzer$1 (ImageAnalysis.java:367)
java.lang.Thread.run (Thread.java:784)`

DanPetras, 2024-04-19:
I believe there is a bug in YuvToRgbConverter.needCreateAllocations function.
The part `bytes.size == yuvBuffer.buffer.capacity()` should actually be `bytes.size != yuvBuffer.buffer.capacity()`

Just to note, there are two YuvToRgbConverter.kt files in this repo, both of them have the same issue.

kailianc, 2024-04-19:
If you are using CameraX ImageAnalysis, you could use this API, it's built on top of libyuv instead of renderscript.
https://developer.android.com/reference/kotlin/androidx/camera/core/ImageAnalysis.Builder#setOutputImageFormat(int)
https://medium.com/androiddevelopers/convert-yuv-to-rgb-for-camerax-imageanalysis-6c627f3a0292

DanPetras, 2024-04-19:
I'm forced to use camera2 api because of AR core.