android/camera-samples #387: Camera2Video: java.lang.IllegalStateException at android.media.MediaRecorder.stop(Native Method)

runaloop, 2021-06-16:
Hello. I have crash while pressing rec button. 

**My steps to reproduce:**

1. Build and install camera2video app on a device
2. Choose any resolution(i have checked few of them, all are crashed)
3. See camera preview and rec button.
4. Press rec button -> crash

**Crash log:**
> E/MediaRecorder: stop called in an invalid state: 4
> E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-3
>     Process: com.android.example.camera2.video, PID: 608
>     java.lang.IllegalStateException
>         at android.media.MediaRecorder.stop(Native Method)
>         at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:297)
>         at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
>         at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
>         at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
> 

**Other details**
git head is: 852767fddd25b0f33e117d811bd5cdb6694b8904
phone: SM-950F

Any thoughts?

zhulg, 2021-06-16:
Same error as #254  you need to press and hold to record

rooton, 2021-07-14:
Yeah, thats google. They can pass interview by solving tons of hard algorithmic tasks, but never able to create good tutorial or even product (excluding google search)

bohdantrotsenko, 2021-08-05:
In my case of Camera2Video crash the last output in console is:

    I/e.camera2.vide: Compiler allocated 4218KB to compile void android.view.ViewRootImpl.performTraversals()
    D/CameraFragment: Recording stopped. Output file: /data/user/0/com.android.example.camera2.video/files/VID_2021_08_05_15_12_11_883.mp4
    I/MediaRecorderJNI: stop
    E/MediaRecorder: stop failed: -1007
    E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-1
        Process: com.android.example.camera2.video, PID: 14156
        java.lang.RuntimeException: stop failed.
            at android.media.MediaRecorder._stop(Native Method)
            at android.media.MediaRecorder.stop(MediaRecorder.java:1472)
            at com.example.android.camera2.video.fragments.CameraFragment$initializeCamera$1$1$2.invokeSuspend(CameraFragment.kt:296)
            at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
            at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:56)
            at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:571)
            at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:738)
            at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:678)
            at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:665)
    I/Process: Sending signal. PID: 14156 SIG: 9

I also cannot see /data/user/ in Total Commander (I have /data/data/... with little of content and /storage/emulated/0). Can this be a root of the problem?

I tried running on Samsung Galaxy Note 9 (SM-N960F/DS), Android 10, security patch level: July 1, 2021.

@owahltinez could you please take a look?

bohdantrotsenko, 2021-08-05:
@serbelga can you please take a look?

bohdantrotsenko, 2021-08-05:
Here's a full log trying to capture on samsung SM-T860:  
(the error is different: `E/MediaRecorder: stop called in an invalid state: 4`)
https://gist.github.com/bohdantrotsenko/bf82e82a701032fb96b10ddd351e90ed

tergel93, 2021-09-01:
> Here's a full log trying to capture on samsung SM-T860:
> (the error is different: `E/MediaRecorder: stop called in an invalid state: 4`)
> https://gist.github.com/bohdantrotsenko/bf82e82a701032fb96b10ddd351e90ed

`
    @Volatile
    private var latestRecordAction: Job? = null

    MotionEvent.ACTION_DOWN -> {
        latestRecordAction = lifecycleScope.launch(Dispatchers.IO) {
            ...
        }
    }
    MotionEvent.ACTION_UP -> lifecycleScope.launch(Dispatchers.IO) {
        latestRecordAction?.join()
        latestRecordAction = null
        ...
    }
`
I added these extra code to assure mediaRecorder is running before we invoke stop. It seems there's some coroutine dispatching issue here.

eduardothiesen, 2021-12-26:
I'm facing the same issue as @bohdantrotsenko. In some cases it causes my device to restart.

Device: Samsung s20+

bohdantrotsenko, 2022-05-20:
the issue is still relevant

bohdantrotsenko, 2022-05-21:
Hi @ggfan  @bbfeechen @jdkoren @temcguir 

_I'm experienced developer in golang/rust/python, but pretty unexperienced in android, so I do make some effort on weekends to solve this myself, but I can't afford spending extra 60h – which is why I need a working sample to build upon_

As this is an official repo,
and you're experienced developers and are also contributors to the repo,
Can you please make sure it doesn't crash?