android/camera-samples #394: fps reduces when image analysis usecase is applied

donpaul120, 2021-07-24:
I'm testing on 

TECNO POP 2F (b1f)
CPU Core-Count = 4
RAM = 1.00G
Internal Storage = 16GB
Screen Resolution = 960 X 480

I'm using the CameraXBasic Example

The camera preview gets laggy only when the ImageAnalysis use case is applied.

I observed that without the imageAnalysis usecase the frame per seconds is approximately 10 on this device even tho  the device will support up to 30fps (Range(lower=30, upper=30)

*Below are the supported fps Range*
Lower => 20, Upper => 20
Lower => 24, Upper => 24
Lower => 5, Upper => 30
Lower => 30, Upper => 30

When ImageAnalsis usecase is applied the fps drops to approximately 3fps

I/BufferQueueProducer: [ImageReader-864x480f23m4-18676-0](this:0x98965000,id:0,api:4,p:18676,c:18676) queueBuffer: fps=2.50 dur=1199.65 max=416.04 min=384.55

And i think this makes the CameraPreview Lag.

Are there any optimization techniques that can be applied to avoid making the preview lag?

rcagala, 2021-10-11:
I experience the same issue on Motorola Moto G4.

Sample: CameraXBasic / Commit: 08efec47db982513bc5faa4c3525c128a6e54c32 ("main" branch)

If I remove the ImageAnalsis use case from bindToLifecycle(), the preview FPS is OK.

ninhkobietnau, 2023-07-24:
same problem. im using cameraX 1.1.0. even if i upgrade to 1.2.0, it still lagging when add imageAnalysis usecase. it just happened with old android version (such as android 7; 8), it terrible

is there any solution ? or this problem is fixed with higher version CameraX