android/camera-samples #403: The Camera2SlowMotion can't run correctly.

BangwenHe, 2021-07-29:
Hi, I am trying to capture high fps(at least 60fps) video and send the YUV data to MediaCodec for hardware h265 codec. But when I run the `Camera2SlowMotion` and select one, it will crash. 
Below are my test info:
- Mi 11 pro
- Android 11

For example, the screen show `1920x1080 120fps`, `1920x1080 240fps`, `1920x1080 480fps`. But I can't open the first two cameras. It throws RunTimeException in  `com.example.android.camera2.slowmo.fragments.CameraFragment$openCamera$2$1.onError`. 

Besides, the recorded video from `1920x1080 480fps` can't be opened by any video player. Log print:
> D/CameraFragment: Recording stopped. Output file: /data/user/0/com.android.example.camera2.slowmo/files/VID_2021_07_29_16_28_59_999.mp4
> I/Timeline: Timeline: Activity_launch_request time:87316611
> D/MediaScannerConnection: Scanned /data/user/0/com.android.example.camera2.slowmo/files/VID_2021_07_29_16_28_59_999.mp4 to null

I also test the app in my own Honor V30 pro. It will crash under any options. Any reply or suggestions can be appreciated.

[bug_report.txt](https://github.com/android/camera-samples/files/6899118/bug_report.txt)

ggfan, 2021-07-30:
Thank you for reporting the issue.  Wondering if you could help to [capture a full bug report](https://developer.android.com/studio/debug/bug-report) right after when the crash happens and share it?

zokica8, 2022-04-06:
Hi. I have the same issue with running this example of capturing a slow motion video on my phone, which is Mi 11T Pro. When I created the bug report for the whole device, I found two lines that could be the reason why this is not working.

`ServiceManager: Permission failure: android.permission.SYSTEM_CAMERA from uid=somenumber pid=somenumber`
`ServiceManager: Permission failure: android.permission.CAMERA_OPEN_CLOSE_LISTENER from uid=somenumber pid=somenumber`

On Android 10 and lower devices it works, from Android 11 it does not work.
The only thing that I found was that from Android 11, the OS introduced the system camera permission, which is assigned at the system level. This is from the Android documentation: 

"On devices running Android 11 or higher, the Android framework supports system cameras, which are camera devices that are visible only to processes with the android.permission.SYSTEM_CAMERA permission and regular camera permissions. The android.permission.SYSTEM_CAMERA permission, introduced in Android 11, has a protection level of system|signature. This means that only apps installed on the system partition with or signed with the same certificate as the system can be granted this permission. Third-party public apps can't access system cameras.

System cameras are useful for device manufacturers that want to implement features that require access to a camera but are also restricted to privileged or system apps.

Because systems apps that are granted the android.permission.SYSTEM_CAMERA permission must also have the [android.permission.CAMERA](https://developer.android.google.cn/reference/android/Manifest.permission#CAMERA) permission, users can choose to revoke normal CAMERA permissions to prevent such an app from accessing the cameras on the device."

Basically what this means is that the only way you can create this kind of an app and run this example is for this app to be a system app, an app that is part of a system, like most Google apps for most Android phones that you can not uninstall from the device, unless you want to root the device.

Is there anybody who has experience in creating these kind of apps?
Also, I found another link that could be useful.

https://blog.csdn.net/KLYAZWS/article/details/119376225

BangwenHe, 2022-04-06:
Thank you for your detailed reply. In the later days after my issue, I just used 60fps recording. With your explanation, I think it can be helpful to the people behind.

closed 2022-04-14 by BangwenHe